mysensors/MySensors #63: Incompatibility between mySensors and softSerial librairies

jaysee, 2015-02-12:
Hello,

I can not create a sensor that use the arduino softserial library, it fails to compile because of "multiple definitions of ..."

**NOTE:** compiling this code using Arduino 1.06

```cpp
// mysensors
#include <SPI.h>
#include <MySensor.h>

#include <SoftwareSerial.h>

MySensor gw;

SoftwareSerial softSerial( 4, 5 );

void setup() {
	softSerial.begin( 1200 );

	gw.begin();
	gw.sendSketchInfo( "softSerial Test Sensor", "0.1" );
}

void loop() {
}
```

**Errors :**
```
SoftwareSerial/SoftwareSerial.cpp.o: In function `__vector_3':
/Applications/Arduino.app/Contents/Resources/Java/libraries/SoftwareSerial/SoftwareSerial.cpp:305: multiple definition of `__vector_3'
MySensors/MyGateway.cpp.o:/Users/jc/Arduino/libraries/MySensors/utility/PinChangeInt.h:563: first defined here
SoftwareSerial/SoftwareSerial.cpp.o: In function `__vector_4':
/Applications/Arduino.app/Contents/Resources/Java/libraries/SoftwareSerial/SoftwareSerial.cpp:312: multiple definition of `__vector_4'
MySensors/MyGateway.cpp.o:/Users/jc/Arduino/libraries/MySensors/utility/PinChangeInt.h:573: first defined here
SoftwareSerial/SoftwareSerial.cpp.o: In function `__vector_5':
/Applications/Arduino.app/Contents/Resources/Java/libraries/SoftwareSerial/SoftwareSerial.cpp:319: multiple definition of `__vector_5'
MySensors/MyGateway.cpp.o:/Users/jc/Arduino/libraries/MySensors/utility/PinChangeInt.h:583: first defined here
```
I could patch the MySensors/utility/PinChangeInt.h to correct the thing but i'm afraid to break something elsewhere, is there somebody that would have a viable solution/patch to propose about that ?

thanks

henrikekblad, 2015-02-12:
What happens if you try the development brach? We have removed the PinChangeInt dependency  over there.

jaysee, 2015-02-12:
Wonderful, it works!

thanks a lot!

closed 2015-02-12 by henrikekblad