mysensors/MySensors #99: Possible missing braces

hmeyers, 2015-04-06:
Hi,

looking at the function MyGateway::setInclusionMode I am wondering if there a missing braces. At least the current ident suggests an intended implementation of:

```
void MyGateway::setInclusionMode(boolean newMode) {
  if (newMode != inclusionMode) {
    inclusionMode = newMode;
    // Send back mode change on serial line to ack command
    serial(PSTR("0;0;%d;0;%d;%d\n"), C_INTERNAL, I_INCLUSION_MODE, inclusionMode?1:0);

    if (inclusionMode) {
      inclusionStartTime = millis();
    }
  }
}
```

henrikekblad, 2015-04-06:
Yes, looks like a bug.

hmeyers, 2015-04-06:
As as suggestion:

Get rid of the if completely as in:

```
void MyGateway::setInclusionMode(boolean newMode) {
  inclusionMode = newMode;
  // Send back mode change on serial line to ack command
  serial(PSTR("0;0;%d;0;%d;%d\n"), C_INTERNAL, I_INCLUSION_MODE, inclusionMode?1:0);
  if (inclusionMode) {
    inclusionStartTime = millis();
  }
}
```

Now if a second inclusion mode request will be received it will extend the inclusion mode time. With the old version it will simply be ignored, possibly confusing the user.

henrikekblad, 2015-04-08:
O, will only fix this in the development-branch for now.

closed 2015-04-08 by henrikekblad