mysensors/MySensors #119: not compiled on arduino due

arturmon, 2015-05-29:
Arduino: 1.5.8 (Windows 8), Board: "Arduino Due (Programming Port)"

In file included from Z:\arduino-1.5.8-new\libraries\DigitalIO/DigitalIO.h:31:0,
                 from MQTTGateway.ino:59:
Z:\arduino-1.5.8-new\libraries\DigitalIO/DigitalPin.h:30:20: fatal error: avr/io.h: No such file or directory
 #include <avr/io.h>
                    ^
compilation terminated.
Ошибка компиляции.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

mahmoud-samy, 2015-07-31:
Check https://github.com/mysensors/Arduino/issues/144

henrikekblad, 2015-07-31:
Unfortunately I don't have any Due to test on. Are you using SoftSPI (see that it fails in DigitalIO). 

In that case is DigitalIO Due compatible?

clearbucketLabs, 2015-08-05:
you don't need a physical arduino due to compile it.

fallberg, 2015-08-05:
Due is not based on AVR so it is (currently) not supported by MySensors.

fallberg, 2015-12-11:
Closing this as it is a known fact that due is based on an unsupported architecture. I suggest opening a new issue with a request for that support or discuss it in the [forum](http://forum.mysensors.org).

closed 2015-12-11 by fallberg