mysensors/MySensors #223: Use portable datatypes in message definition

Yveaux, 2015-10-08:
MyMessage, used for message data transfer, is defined as follows:
```
	union {
		uint8_t bValue;
		unsigned long ulValue;
		long lValue;
		unsigned int uiValue;
		int iValue;
		struct { // Float messages
			float fValue;
			uint8_t fPrecision;   // Number of decimals when serializing
		};
		struct {  // Presentation messages
			uint8_t version; 	  // Library version
   		    uint8_t sensorType;   // Sensor type hint for controller, see table above
		};
		char data[MAX_PAYLOAD + 1];
	} __attribute__((packed));
```

The (unsigned) int types are non-portable when going from 8 bit AVR architecture to 32 bit (e.g. ESP8266, Cortex-M), which is a problem when architectures are mixed within a network.
On AVR an (unsigned) int is 16bits, on ESP it is 32 bits.
On AVR a double is 32bits (equal to float), on ESP it is 64 bits.

This probably lead to a problem described in the forum: http://forum.mysensors.org/topic/2118/esp8266-wifi-gateway-issue

For portability use uint16_t and int16_t for ints.
For consistency also use uint32_t and int32_t for longs, uint8_t and int8_t for chars.

--------------------
Background

Following simple sketch can be run on different targets to identify the sizes of their datatypes:

```
#define SIZEOF(x) \
  Serial.print("sizeof(" #x ")="); Serial.println(sizeof(x));

void setup() {
  Serial.begin(115200);
  SIZEOF(char);
  SIZEOF(int);
  SIZEOF(short);
  SIZEOF(long);
  SIZEOF(float);
  SIZEOF(double);
  SIZEOF(uint8_t);
  SIZEOF(uint16_t);
  SIZEOF(uint32_t);
}

void loop() {
}
```

Output on ATMega328 (8bit)
```
sizeof(char)=1
sizeof(int)=2
sizeof(short)=2
sizeof(long)=4
sizeof(float)=4
sizeof(double)=4
sizeof(uint8_t)=1
sizeof(uint16_t)=2
sizeof(uint32_t)=4
```

Output on ESP8266 (32bit)
```
sizeof(char)=1
sizeof(int)=4            <-- Problematic
sizeof(short)=2
sizeof(long)=4
sizeof(float)=4
sizeof(double)=8         <-- Problematic, but not used in MyMessage
sizeof(uint8_t)=1
sizeof(uint16_t)=2
sizeof(uint32_t)=4
```

fallberg, 2015-10-08:
I agree. For any data structure "leaving" a core, all types must be explicitly expressed in terms of size.

henrikekblad, 2015-10-08:
Good investigation.

PR #224 (2015-10-08): Use sdtint for in MyMessage.
Fixes data type length problem when messages travels between 8/16/32 bit
architectures. 

close #223
commit 70b87162 (2015-10-08): Use sdtint for in MyMessage.  

Fixes data type length problem when messages travels between 8/16/32 bit
architectures. 

close #223
closed 2015-10-08 by henrikekblad