mysensors/MySensors #225: cloning is on development branch by default

empierre, 2015-10-10:
and this is puzzling...

people should be warned to use : 
git clone https://github.com/mysensors/Arduino.git Arduino-MySensors -b master

fallberg, 2015-10-10:
This is Intentional and documented behaviour: http://forum.mysensors.org/topic/330/how-to-contribute-code-to-the-mysensors-project/25

empierre, 2015-10-10:
agree for contributors, but for the rest of the workld in API & sources we
propose the cloning without warning on this, thus I got in the trap and
could not compile anymore when I just wanted to update my main branch for
use and not for dev

fallberg, 2015-10-10:
Yes for sure, the comment could be extended for those who wants to use master with git (in contrast to just download the zip for the release in question on master. Plan now is to only use master for releases, so there should always he an option to download a zip instead of pulling (although pulling is convenient).
But it is typically not common to re-clone over an existing project. It is only when you want a new project you should clone. And yes, now if you clone without stating exactly what you want, you end up on development. It has been discussed back and forth which behaviour is most desirable and it was concluded that most people working with git do it in order to do some form of development, typically resulting in a pull request. And if default branch is master, default pr target is also master, and the build server do not permit pr:s to master (intentionally) so by having development as default the process of creating pr:s is simplified at the expense of the default clone behaviour. @henrikekblad perhaps can make some statement on the forum or update a topic on working with MySensors through git to clarify this.

empierre, 2015-10-10:
Hello,

  I first did a pull and could not use it again... I thought I was on
main...

  then I did clone it again to have the same result, this is why I say so.
If you stay on the clone dev version, please remove the clone link in the
"getting the API"

henrikekblad, 2015-10-10:
If you've cloned to your local computer you'll get all branches. To switch brach to "master" you simply type:

```> git checkout master```

empierre, 2015-10-10:
yes I now know, I just pointed that someone that did get the github on
http://www.mysensors.org/download/ may get lost (as I did..)

henrikekblad, 2015-10-10:
Yes, but the "Download"-zip link points to master branch.

empierre, 2015-10-10:
I would not download the zip for if there are updates I wouldn't know...
the git pull tells me that ;-)

fallberg, 2015-10-10:
It's always good practice to specify branch when pulling. Especially now when the most recent version of git actually changed that behaviour.

fallberg, 2015-10-10:
Also, defaults for remotes and tracking branches are up to each user to configure locally, so all git savvy developers/users can set up their clones just as they want them.

empierre, 2015-10-10:
I agree, but we have soooo many people with different background around ot
that may come we need to help them jumlpstart, not scratch their heads ;-)

fallberg, 2015-10-10:
Then I'd suggest "download". Linus never intended git to be "user friendly". He wanted it "fast". :)

fallberg, 2015-10-10:
But of course the instructions on how to download the stuff should be correct. And reasonable startup info for those who want to contribute should be available and up to date.

empierre, 2015-10-10:
I agree ;-)

closed 2015-11-15 by henrikekblad