mysensors/MySensors #272: MyProtocolMySensors.cpp - protocolParse - check if value is null

ciprianpop3scu, 2015-11-29:
in https://github.com/mysensors/Arduino/blob/development/libraries/MySensors/core/MyProtocolMySensors.cpp#L94

should be checked if value is not null, because in the case of a message without payload (such as internal) it will not go to "case 5: // Variable value" and value will remain NULL (char *str, *p, *value=NULL;)

Thanks

henrikekblad, 2015-11-29:
Thanks for reporting. Are you by any chance on ESP8266 arch?

I've created a PR to fix NULL check here (will be merged after build server has finished):
https://github.com/mysensors/Arduino/pull/273

Maybe you can verify this on your hardware?

closed 2015-11-30 by henrikekblad

ciprianpop3scu, 2015-12-08:
Sorry for the delay! I can confirm that it is working  for ESP8266(tested with ESP8266Gateway):
0;0;3;0;9;Client 0 connected
0;0;3;0;9;Client 0: 0;0;3;0;2;
0;0;3;0;9;Client 0: 0;0;3;0;2;Get Version
This is where, before the fix, it was chrashing. Now it is working. 
Thanks.

commit 6d97702b (2015-11-30): Fix #272 - NULL argument check in MyMessage::set(char *)