mysensors/MySensors #275: ESP8266 Arduino Core 2.0.0 and Multiple definitions of abort()

mannkind, 2015-12-01:
### Steps to Replicate
* Upgrade the [ESP8266 Arduino core](https://github.com/esp8266/Arduino) to 2.0.0
* Attempt to compile the example Gateway ESP8266 MQTT Client Sketch.

### What Happens
Compilation fails w/the error `multiple definition of 'abort'`

### Current Work-Arounds
* Revert to the 1.6.5 version of the  ESP8266 Arduino core
* Rename [this function and subsequent calls](https://github.com/mysensors/Arduino/blob/development/libraries/MySensors/core/MyMainESP8266.cpp#L46).   
  **Note:** I have not confirmed this actually *functions* correctly, only that it compiles.

henrikekblad, 2015-12-01:
Thanks @mannkind, 

Saw reports on the forum. I will have a look right away when I'm back home from a trip.

henrikekblad, 2015-12-05:
fixed

closed 2015-12-05 by henrikekblad