mysensors/MySensors #276: DallasTemperature Scketch

redvoodoochild, 2015-12-02:
Hi.
Excuse my poor english.

My setup : Arduino 1.6.0 on archlinux / Arduino Uno
Dallas Sensor connected on the serial gateway (common scketch)

The sctech complie, but all the intems on the Void Presentation doesn't work.
If I put this items on the void setup, it's work : my sensors are recognized by my controller via the gateway.

fallberg, 2015-12-11:
Hm, you mean that your sensor is not properly presented to the gateway? On development branch there are numerous examples that use the presentation() function so you should be able to find a lot of examples to be inspired of when deciding your setup.

redvoodoochild, 2015-12-12:
Yes. I use the examples for my scketch. I've done a scketch with a relay sensor and somme dallas temperature.
When i put the the presentation of the dallas temperature on the void presentation() they are not presented to the gateway (for the relay it works). I must put the dallas presentation on the void setup() for them to be presentated. I will post my scketch and the serial monitor (but i must disconnect them from my controller...)

fallberg, 2015-12-12:
And your Dallas node is running 1.6.0-beta? (development branch)

redvoodoochild, 2015-12-20:
Yes.
This is my working scketch :
```
  /**
 * The MySensors Arduino library handles the wireless radio link and protocol
 * between your home built sensors/actuators and HA controller of choice.
 * The sensors forms a self healing radio network with opjjtional repeaters. Each
 * repeater and gateway builds a routing tables in EEPROM which keeps track of the
 * network topology allowing messages to be routed to nodes.
 *
 * Created by Henrik Ekblad <henrik.ekblad@mysensors.org>
 * Copyright (C) 2013-2015 Sensnology AB
 * Full contributor list: https://github.com/mysensors/Arduino/graphs/contributors
 *
 * Documentation: http://www.mysensors.org
 * Support Forum: http://forum.mysensors.org
 *
 * This program is free software; you can redistribute it and/or
 * modify it under the terms of the GNU General Public License
 * version 2 as published by the Free Software Foundation.
 *
 *******************************
 *
 * DESCRIPTION
 * The ArduinoGateway prints data received from sensors on the serial link. 
 * The gateway accepts input on seral which will be sent out on radio network.
 *
 * The GW code is designed for Arduino Nano 328p / 16MHz
 *
 * Wire connections (OPTIONAL):
 * - Inclusion button should be connected between digital pin 3 and GND  
 * - RX/TX/ERR leds need to be connected between +5V (anode) and digital pin 6/5/4 with resistor 270-330R in a series
 *
 * LEDs (OPTIONAL):
 * - To use the feature, uncomment MY_LEDS_BLINKING_FEATURE in MyConfig.h
 * - RX (green) - blink fast on radio message recieved. In inclusion mode will blink fast only on presentation recieved
 * - TX (yellow) - blink fast on radio message transmitted. In inclusion mode will blink slowly
 * - ERR (red) - fast blink on error during transmission error or recieve crc error 
 * 
 */

// Enable debug prints to serial monitor
#define MY_DEBUG 

// Enable and select radio type attached
//#define MY_RADIO_NRF24  <= Je n'ai toujours pas branché d'antennes (en attente matos pour créer mes nodes
//#define MY_RADIO_RFM69

// Set LOW transmit power level as default, if you have an amplified NRF-module and
// power your radio separately with a good regulator you can turn up PA level. 
#define MY_RF24_PA_LEVEL RF24_PA_LOW

// Enable serial gateway
#define MY_GATEWAY_SERIAL

// Flash leds on rx/tx/err
#define MY_LEDS_BLINKING_FEATURE
// Set blinking period
#define MY_DEFAULT_LED_BLINK_PERIOD 300

// Inverses the behavior of leds
//#define MY_WITH_LEDS_BLINKING_INVERSE

// Enable inclusion mode
#define MY_INCLUSION_MODE_FEATURE
// Enable Inclusion mode button on gateway
#define MY_INCLUSION_BUTTON_FEATURE

// Inverses behavior of inclusion button (if using external pullup)
//#define MY_INCLUSION_BUTTON_EXTERNAL_PULLUP
[... 334 lines cut ...]
     // Write some debug info
     Serial.print("Incoming change for sensor:");
     Serial.print(message.sensor);
     Serial.print(", New status: ");
     Serial.println(message.getBool());
   } 
}
```
The serial monitor :
```
0;0;3;0;9;Starting gateway (1.6.0-beta)
0;0;3;0;14;Gateway startup complete.
0;255;3;0;11;Relay + Temp
0;255;3;0;12;1.0
0;1;0;0;3;
0;2;0;0;3;
0;0;3;0;9;Init complete, id=0, parent=0, distance=0
0;0;3;0;5;1
0;0;3;0;5;0
0;0;3;0;5;1
```

Hope will help.

Best regards.

fallberg, 2015-12-25:
Hm, I'm on phone so I have trouble getting a good overview but I can't seem to see anything odd. @henrikekblad any ideas?

henrikekblad, 2015-12-25:
I think you should download the latest from the dev branch.. The call order of setup and presentation has changed which might help in your situation (where you initialise Dallas in setup and do the actual presentation in presentation().

redvoodoochild, 2016-01-06:
Well.
I've downloaded the lastest dev branch, I've erased the nano's eeprom, and all works well !!
Thanks.

closed 2016-01-06 by redvoodoochild