mysensors/MySensors #285: Signing, MYSBootloader, OTA firmware uploads

mannkind, 2015-12-10:
I just flashed the [MYSBootloader](https://github.com/mysensors/MySensorsBootloaderRF24) to nearly all of my nodes. 

I've had great success w/the OTA firmware upload *except* on the handful of nodes that have signing enabled via `#define MY_SIGNING_REQUEST_SIGNATURES`. I can confirm the same node, same firmware (minus `#define MY_SIGNING_REQUEST_SIGNATURES`) works … until a firmware with signing enabled is uploaded.

With `#define MY_SIGNING_REQUEST_SIGNATURES` enabled, this is what the node startup looks like on the Gateway. Notice the nonce timeouts at the bottom.

```
0;0;3;0;9;read and forward: 10-10-255 s=255,c=3,t=7,pt=0,l=0,sg=0
0;0;3;0;9;send: 0-0-10-10 s=255,c=3,t=8,pt=1,l=1,sg=0,st=ok:0
0;0;3;0;9;read: 10-10-0 s=255,c=4,t=0,pt=6,l=10,sg=0:0500010010036A540101
0;0;3;0;9;Sending message on topic: mysensors_rx/10/255/4/0/0
0;0;3;0;9;Message arrived on topic: mysensors_tx/10/255/4/0/1
0;0;3;0;9;send: 0-0-10-10 s=255,c=3,t=16,pt=0,l=0,sg=0,st=ok:
0;0;3;0;9;read: 10-10-0 s=255,c=4,t=0,pt=6,l=10,sg=0:0500010010036A540101
0;0;3;0;9;Sending message on topic: mysensors_rx/10/255/4/0/0
0;0;3;0;9;Message arrived on topic: mysensors_tx/10/255/4/0/1
0;0;3;0;9;send: 0-0-10-10 s=255,c=3,t=16,pt=0,l=0,sg=0,st=ok:
0;0;3;0;9;read: 10-10-0 s=255,c=4,t=0,pt=6,l=10,sg=0:0500010010036A540101
0;0;3;0;9;Sending message on topic: mysensors_rx/10/255/4/0/0
0;0;3;0;9;Message arrived on topic: mysensors_tx/10/255/4/0/1
0;0;3;0;9;send: 0-0-10-10 s=255,c=3,t=16,pt=0,l=0,sg=0,st=ok:
0;0;3;0;9;read: 10-10-0 s=255,c=4,t=0,pt=6,l=10,sg=0:0500010010036A540101
0;0;3;0;9;Sending message on topic: mysensors_rx/10/255/4/0/0
0;0;3;0;9;Message arrived on topic: mysensors_tx/10/255/4/0/1
0;0;3;0;9;send: 0-0-10-10 s=255,c=3,t=16,pt=0,l=0,sg=0,st=ok:
0;0;3;0;9;read: 10-10-0 s=255,c=4,t=0,pt=6,l=10,sg=0:0500010010036A540101
0;0;3;0;9;Sending message on topic: mysensors_rx/10/255/4/0/0
0;0;3;0;9;Message arrived on topic: mysensors_tx/10/255/4/0/1
0;0;3;0;9;send: 0-0-10-10 s=255,c=3,t=16,pt=0,l=0,sg=0,st=ok:
0;0;3;0;9;read: 10-10-0 s=255,c=3,t=15,pt=2,l=2,sg=0:1
0;0;3;0;9;send: 0-0-10-10 s=255,c=3,t=15,pt=4,l=4,sg=0,st=ok:0
0;0;3;0;9;read: 10-10-0 s=255,c=0,t=17,pt=0,l=10,sg=0:1.6.0-beta
0;0;3;0;9;Sending message on topic: mysensors_rx/10/255/0/0/17
0;0;3;0;9;read: 10-10-0 s=255,c=3,t=6,pt=1,l=1,sg=0:0
0;0;3;0;9;Sending message on topic: mysensors_rx/10/255/3/0/6
0;0;3;0;9;Message arrived on topic: mysensors_tx/10/255/3/0/6
0;0;3;0;9;send: 0-0-10-10 s=255,c=3,t=16,pt=0,l=0,sg=0,st=ok:
0;0;3;0;9;read: 10-10-0 s=255,c=3,t=17,pt=6,l=25,sg=0:0149901E44791163512E42160B844B615405D2CAE928D4029
0;0;3;0;9;Signing backend: ATSHA204Soft
0;0;3;0;9;Message to process: 000A0E0306FF49
0;0;3;0;9;Current nonce: 0149901E44791163512E42160B844B615405D2CAE928D402A9AAAAAAAAAAAAAA
0;0;3;0;9;HMAC: A64E8650FE3345FA13FE4BC4084DBB134A3993346AE8D9A9B6A1FE9C25172AA7
0;0;3;0;9;Signature in message: 014E8650FE3345FA13FE4BC4084DBB134A3993346AE8D9A9
0;0;3;0;9;send: 0-0-10-10 s=255,c=3,t=6,pt=0,l=1,sg=1,st=ok:I
0;0;3;0;9;send: 0-0-10-10 s=255,c=3,t=6,pt=0,l=1,sg=1,st=ok:I
0;0;3;0;9;nonce tmo
0;0;3;0;9;nonce tmo
0;0;3;0;9;nonce too
0;0;3;0;9;nonce tmo
```

fallberg, 2015-12-10:
OTA and signing is not supported for mysbooloader. The bootloader does not handle signing. It should work with DualOptiboot but I have not confirmed that yet.

mannkind, 2015-12-10:
Ah, interesting. Perhaps I overlooked an update, but the introduction post about signing seemed to indicate that, "OTA firmware updates are transmitted unsigned" and since the MYSBootlosder doesn't know about singing, everything would work.

fallberg, 2015-12-10:
So would I, but it appears that is not the case. I suspect mysbooloader transmits other types of messages as well that normally are supposed to be signed. I am afraid I don't know the communication details when doing OTA through mysbooloader so I am not sure what's going on in that case. @tekka might know where a patch is needed.

tekka007, 2015-12-10:
Yes, MYSBootloader 1.1 does not inform the GW about it's signing preference. This has been addressed in 1.4. @mannkind I can send you 1.4 for testing.

mannkind, 2015-12-10:
@tekka007 sure! I'd love to help. Is there anything specific you'd like me to do? Aside from trying it on nodes w/signing enabled? I'm using the default setup - ch. 76, ce 9, csn 10.

tekka007, 2015-12-10:
I'll send you a PM

closed 2015-12-11 by fallberg