mysensors/MySensors #293: preBeginn Function

fbeek, 2015-12-11:
Hey guys,

at the moment i am working on an sensor node with an MRFC522 RFID reader beside the NRF24 on the SPI Bus. 

With Version 1.5.1 it was possible to run code before the radio init function ( gw.begin(); )so that you could do something like this to stop the rfid reader on the SPI Bus so that there where no interferences on the bus with the NRF Module:
```
  pinMode(RST_PIN, OUTPUT);     
  digitalWrite(RST_PIN, LOW);
  pinMode(SS_PIN, OUTPUT);     
  digitalWrite(SS_PIN, LOW);
  gw.begin();
```

In Version 1.6 you run the setup() function after the radio init in your _beginn function so that above mentioned code comes to late and the radio init fails.

Is it possible that you could implement something like the loop or setup function that comes before? So that we can do such thing before?

I would implement this by myself but my CFu is not very good. 

// Initialize library and handle sketch functions like we want to
```
int main(void) {
	init();
	#if defined(USBCON)
		USBDevice.init();
    		USBDevice.attach();
	#endif
          if (preBegin) preBegin();  <------
	_begin(); // Startup MySensors library

	for(;;) {
		_process();  // Process incoming data
		if (loop) loop(); // Call sketch loop
		if (serialEventRun) serialEventRun();
	}
	return 0;
}
```

Regards 

Florian

PressY4EKG, 2016-01-27:
i know this is a little old, but in the development branch the first execution is in _begin().
so all i did was move the calling of setup to before transportInit() gets started. id imagine there is a reason as to why they made it work this way, but i moved the setup setup call without anything bad happening.

henrikekblad, 2016-02-06:
Some of the examples request data from the controller in the begin-method. If transportInit() is called after begin the will fail. 

A preBegin might be a better option where sensor initializations can happen.

PR #386 (2016-02-06): Add before() for initialisations before radio hogs SPI
Fix #293
commit 469cbb7d (2016-02-06): Add before() for initialisations before radio hogs SPI

Fix #293
closed 2016-02-06 by fallberg