mysensors/MySensors #296: [Security] signing bypass flaw

tekka007, 2015-12-12:
Messages with ACK-bit enabled bypass signing verification.

fallberg, 2015-12-12:
Intended behaviour in the context that acks are not supposed to be signed but it was erroneously assumed that acks were simply sent to the caller without further processing. 
The fix for this will be to require acks to be signed (if the receiver of the ACK require this).

tekka007, 2015-12-12:
Signing ACK is not a bad idea, however, it won't necessarily solve the issue here. The fixes are (1) to not process ACK messages in the library and (2) verify ACK-bit in the callback function before further processing (see SecureActuator sketch). 
We should eventually consider a different way of ACKing a message, i.e. via a designated I_type + full hash of the original message. Let's have a chat about that.

Avamander, 2015-12-13:
(I just noticed many issues have [SECURITY] in the title, wouldn't it be easier to create a new issue tracker tag and tag the posts with "Security issue"?)

fallberg, 2015-12-13:
It's just for me to keep track of my backlog. Generally this feature is not used by the MySensors community.

fallberg, 2015-12-13:
It's not only issues related to security. It is also various work items.

fallberg, 2016-01-02:
Secure actuator has been "secured" in the sense that ACKs are now not accepted as data carrying messages. See 62f522. I consider the issue fixed with this.

closed 2016-01-02 by fallberg