mysensors/MySensors #322: Int value gets the rest of a float value when you change betw. reading  Integer and Float

maxtox, 2016-01-04:
for example:
1. sketch  reads with getInt a value 10 sent from Server

2. temperature sensor sends a value 47.28 to the server

3. sketch  reads with getInt a value 1028 (10 was sent from Server)

OR

1. sketch  reads with getFloat a value 0 or -20.20 sent from Server

2. temperature sensor sends a value 1.28 to the server

3. sketch  reads with getFloat a value 1.28(0 was sent from Server)
check here:
```
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/13/1/0/3
13
-20.20
0;255;3;0;9;Sending message on topic: DOino<>MQTT/0/58/1/0/0
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/58/1/0/0
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/13/1/0/3
im pwm
13
1.28
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/13/1/0/3
13
-20.20
````
OR

1. sketch  reads with getFloat a value 0 or -20.20 sent from Server

3. sketch  reads with getFloat a value 20.20(0 was sent from Server)
check here:
````
13
-20.20
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/13/1/0/3
13
20.20
0;255;3;0;9;Sending message on topic: DOino<>MQTT/0/58/1/0/0
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/58/1/0/0
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/13/1/0/3
13
-20.20
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/13/1/0/3
13
20.20
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/13/1/0/3
13
-20.20
````

could you please check! can send you my sketch if you like!

henrikekblad, 2016-01-04:
Please try the development branch.

maxtox, 2016-01-04:
Already did... The latest

henrikekblad, 2016-01-04:
Enable debug prints in both sensor and gateway and post your logs.

maxtox, 2016-01-04:
here we go:

server:
````
mqtt.0	2016-01-04 17:24:22	info	mqtt.0 Send to client [mqtt_856eed7a.7a911] 'mqtt/0/DOino<>MQTT/0/11/1/0/3': 100
mqtt.0	2016-01-04 17:24:22	info	mqtt.0 Send to client [DOino_1] 'DOino<>MQTT/0/11/1/0/3': 100
mqtt.0	2016-01-04 17:24:22	debug	inMem message * mqtt.0.DOino<>MQTT.0.11.1.0.3
mqtt.0	2016-01-04 17:24:22	info	mqtt.0 Client [mqtt_856eed7a.7a911] _publishes 'mqtt.0.DOino<>MQTT.0.11.1.0.3'(number): 100
mqtt.0	2016-01-04 17:24:18	info	mqtt.0 Send to client [mqtt_856eed7a.7a911] 'mqtt/0/DOino<>MQTT/0/58/1/0/0': 40.95
mqtt.0	2016-01-04 17:24:18	info	mqtt.0 Send to client [DOino_1] 'DOino<>MQTT/0/58/1/0/0': 40.95
mqtt.0	2016-01-04 17:24:18	debug	inMem message * mqtt.0.DOino<>MQTT.0.58.1.0.0
mqtt.0	2016-01-04 17:24:18	info	mqtt.0 Client [DOino_1] _publishes 'mqtt.0.DOino<>MQTT.0.58.1.0.0'(number): 40.95
mqtt.0	2016-01-04 17:24:17	info	mqtt.0 Send to client [mqtt_856eed7a.7a911] 'mqtt/0/DOino<>MQTT/0/11/1/0/3': 0
mqtt.0	2016-01-04 17:24:17	info	mqtt.0 Send to client [DOino_1] 'DOino<>MQTT/0/11/1/0/3': 0
mqtt.0	2016-01-04 17:24:17	debug	inMem message * mqtt.0.DOino<>MQTT.0.11.1.0.3
mqtt.0	2016-01-04 17:24:17	info	mqtt.0 Client [mqtt_856eed7a.7a911] _publishes 'mqtt.0.DOino<>MQTT.0.11.1.0.3'(number): 0
mqtt.0	2016-01-04 17:24:16	info	mqtt.0 Send to client [mqtt_856eed7a.7a911] 'system/adapter/admin/0/uptime': 116978
mqtt.0	2016-01-04 17:24:16	debug	inMem message * system.adapter.admin.0.uptime
mqtt.0	2016-01-04 17:24:16	info	mqtt.0 Send to client [mqtt_856eed7a.7a911] 'system/adapter/admin/0/memHeapUsed': 17.09
mqtt.0	2016-01-04 17:24:16	debug	inMem message * system.adapter.admin.0.memHeapUsed
mqtt.0	2016-01-04 17:24:16	debug	inMem message * system.adapter.admin.0.memHeapTotal
mqtt.0	2016-01-04 17:24:16	debug	inMem message * system.adapter.admin.0.memRss
mqtt.0	2016-01-04 17:24:16	info	mqtt.0 Send to client [mqtt_856eed7a.7a911] 'system/adapter/node-red/0/uptime': 112555
mqtt.0	2016-01-04 17:24:16	debug	inMem message * system.adapter.node-red.0.uptime
mqtt.0	2016-01-04 17:24:16	debug	inMem message * system.adapter.admin.0.connected
mqtt.0	2016-01-04 17:24:16	info	mqtt.0 Send to client [mqtt_856eed7a.7a911] 'system/adapter/node-red/0/memHeapUsed': 17.22
mqtt.0	2016-01-04 17:24:16	debug	inMem message * system.adapter.node-red.0.memHeapUsed
mqtt.0	2016-01-04 17:24:16	debug	inMem message * system.adapter.admin.0.alive
mqtt.0	2016-01-04 17:24:16	debug	inMem message * system.adapter.node-red.0.memHeapTotal
mqtt.0	2016-01-04 17:24:16	debug	inMem message * system.adapter.node-red.0.memRss
mqtt.0	2016-01-04 17:24:16	debug	inMem message * system.adapter.node-red.0.connected
mqtt.0	2016-01-04 17:24:16	debug	inMem message * system.adapter.node-red.0.alive
mqtt.0	2016-01-04 17:24:13	debug	mqtt.0 Client [DOino_1] pingreq
mqtt.0	2016-01-04 17:24:12	info	mqtt.0 Send to client [mqtt_856eed7a.7a911] 'mqtt/0/DOino<>MQTT/0/11/1/0/3': 100
mqtt.0	2016-01-04 17:24:12	info	mqtt.0 Send to client [DOino_1] 'DOino<>MQTT/0/11/1/0/3': 100
mqtt.0	2016-01-04 17:24:12	debug	inMem message * mqtt.0.DOino<>MQTT.0.11.1.0.3
mqtt.0	2016-01-04 17:24:12	info	mqtt.0 Client [mqtt_856eed7a.7a911] _publishes 'mqtt.0.DOino<>MQTT.0.11.1.0.3'(number): 100
````

NOde:

````
10084.00
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/11/1/0/3

11
84.00
0;255;3;0;9;Sending message on topic: DOino<>MQTT/0/58/1/0/0
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/58/1/0/0
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/11/1/0/3

11
10039.00
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/11/1/0/3

11
39.00
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/11/1/0/3

11
10039.00
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/11/1/0/3

11
39.00
0;255;3;0;9;Sending message on topic: DOino<>MQTT/0/58/1/0/0
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/58/1/0/0
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/11/1/0/3

11
10030.00
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/11/1/0/3

11
30.00
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/11/1/0/3

11
10030.00
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/11/1/0/3

11
30.00
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/11/1/0/3

11
10030.00
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/11/1/0/3

11
30.00
````

henrikekblad, 2016-01-04:
Do you have sensors directly attached to the mqtt-gateway?
You seem to have same prefix for incoming/outgoing messages.. which means they will be loop-backed from the mqtt-broker.

maxtox, 2016-01-04:
yes... OUT and IN is the same! Is it not allowed?

I did not want to devide it in the server... wanted to have i in the same tree! Is there a no way around to "clean" values before writing a new value?

maxtox, 2016-01-04:
up til now, using iobroker with mqtt server, i did not have the issue with "LOOPING BACK"... so it was not an issue yet :-)

ok... devided not In and OUT...

but SAME ISSUE see log:`(when i start to send a second value (ID 58) parallel to the (ID 13) it starting to get crazy!

````
13
0.00

0;255;3;0;9;Message arrived on topic: DOino<>MQTT IN/0/13/1/0/3

13
100.00

0;255;3;0;9;Message arrived on topic: DOino<>MQTT IN/0/13/1/0/3

13
0.00

0;255;3;0;9;Message arrived on topic: DOino<>MQTT IN/0/13/1/0/3

13
100.00

0;255;3;0;9;Message arrived on topic: DOino<>MQTT IN/0/13/1/0/3

13
0.00

0;255;3;0;9;Message arrived on topic: DOino<>MQTT IN/0/13/1/0/3

13
100.00

0;255;3;0;9;Message arrived on topic: DOino<>MQTT IN/0/13/1/0/3

13
0.00

0;255;3;0;9;Sending message on topic: DOino<>MQTT OUT/0/58/1/0/0
0;255;3;0;9;Message arrived on topic: DOino<>MQTT IN/0/13/1/0/3

13
100.00

0;255;3;0;9;Message arrived on topic: DOino<>MQTT IN/0/13/1/0/3

13
0.00

0;255;3;0;9;Message arrived on topic: DOino<>MQTT IN/0/13/1/0/3

13
100.00

0;255;3;0;9;Message arrived on topic: DOino<>MQTT IN/0/13/1/0/3

13
0.00

0;255;3;0;9;Message arrived on topic: DOino<>MQTT IN/0/58/1/0/0
0;255;3;0;9;Message arrived on topic: DOino<>MQTT IN/0/13/1/0/3

13
100368.00

0;255;3;0;9;Message arrived on topic: DOino<>MQTT IN/0/58/1/0/0
0;255;3;0;9;Message arrived on topic: DOino<>MQTT IN/0/13/1/0/3

13
0.26

0;255;3;0;9;Message arrived on topic: DOino<>MQTT IN/0/58/1/0/0
0;255;3;0;9;Sending message on topic: DOino<>MQTT OUT/0/58/1/0/0
0;255;3;0;9;Message arrived on topic: DOino<>MQTT IN/0/13/1/0/3

13
100368.00

0;255;3;0;9;Message arrived on topic: DOino<>MQTT IN/0/58/1/0/0
0;255;3;0;9;Message arrived on topic: DOino<>MQTT IN/0/13/1/0/3

13
0.26

0;255;3;0;9;Message arrived on topic: DOino<>MQTT IN/0/58/1/0/0
0;255;3;0;9;Message arrived on topic: DOino<>MQTT IN/0/13/1/0/3

13
100368.00

0;255;3;0;9;Message arrived on topic: DOino<>MQTT IN/0/58/1/0/0
0;255;3;0;9;Message arrived on topic: DOino<>MQTT IN/0/13/1/0/3

13
0.26

0;255;3;0;9;Message arrived on topic: DOino<>MQTT IN/0/58/1/0/0
0;255;3;0;9;Message arrived on topic: DOino<>MQTT IN/0/13/1/0/3

13
100368.00

0;255;3;0;9;Sending message on topic: DOino<>MQTT OUT/0/58/1/0/0
0;255;3;0;9;Message arrived on topic: DOino<>MQTT IN/0/58/1/0/0
````

henrikekblad, 2016-01-04:
Post your sketch also.

maxtox, 2016-01-04:
````

`void receive(const MyMessage &message)
{
  int childPin = message.sensor;
  //  Serial.println(message.getBool());

  for (int i = 0; i<iomodus_count; i++) ////search All Sensor Variables and set them
  {
    MyMessage msg(i, iomodus[i].variableType);
    while (iomodus[i].sensorType == S_UNUSED) { i++; }  //do not waste time to check UNUSED pins

                              //********************************IF VAR1 = Temp OFFSET existst*************************************** 
    if (message.type == V_VAR1)
    {
      if ((childPin == i) && (iomodus[i].sensorType == S_TEMP) && (iomodus[i].sensorVersion == "MAX31855"))
      {
        Temp_offset[i] = message.getFloat();
      }
    }
    if (message.type == V_VAR1)
    {
      if ((childPin == i) && (iomodus[i].sensorType == S_TEMP) && (iomodus[i].sensorVersion == "NTC"))
      {
        Temp_offset[i] = message.getFloat();
      }
    }
    //********************************IF VAR2 = LPF (Low Pass Filter existst*************************************** 
    if (message.type == V_VAR2)
    {
      //***************Low Pass Filter**************
      if ((childPin == i) && (iomodus[i].sensorType == S_TEMP) && (iomodus[i].sensorVersion == "MAX31855") && (message.getFloat() <= 1.0))
      {
        ktype_LPF_rate[i] = message.getFloat();
      }
    }
    //********************************IF V_STATUS or V_LIGHT existst*************************************** 
    if (message.type == V_STATUS)
    {
      if ((childPin == i) && (iomodus[i].sensorType == S_BINARY) && (iomodus[i].sensorVersion == "RELAY"))
      {
        if (relstate[i] != message.getBool())
        {
          digitalWrite(i, message.getBool() ? RELAY_ON : RELAY_OFF);
          relstate[i] = message.getBool();
          // Store state in eeprom
          saveState(i, message.getBool());
        }
      }
    }
    //********************************IF V_PERCENTAGE or V_DIMMER existst*************************************** 
    if ((childPin == i) && (message.type == V_PERCENTAGE) && (message.getFloat() != last_int_value[i]))
    {
      Serial.println("im pwm");
      Serial.println(i);
      Serial.println(message.getFloat());
      //***************PWM Rate 0 to 100 (%)**************  
      if ((childPin == i) && (iomodus[i].sensorVersion == "PWM") && (i <= 13))
      {
        Serial.println("im digital");
        pinMode(i, OUTPUT);
        analogWrite(i, message.getFloat()*2.55); // Set Value 0-255
        last_int_value[i] = message.getFloat();
      }
      //http://www.instructables.com/id/Analog-Output-Convert-PWM-to-Voltage/
      if ((childPin == i) && (iomodus[i].sensorVersion == "PWM_5V") && (i >= 54) && (i <= 69))
      {
        Serial.println("im analog");
        pinMode(i, OUTPUT);
        analogWrite(i, message.getFloat()*2.55); // Set Value 0-255
        last_int_value[i] = message.getFloat();
      }

    }

  }

}````

henrikekblad, 2016-01-04:
Hmm.. A bit hard to read out what you try to archive here...

But what happens if you read payload with getString instead and convert it to a float? 

Shouldn't these type of comparison be done with strcmp?
iomodus[i].sensorVersion == "PWM_5V"

maxtox, 2016-01-04:
sorry for question...

how do i do it quickly?

maxtox, 2016-01-04:
ok...did it with
String s = String(message.getString());
  int get_i= s.toInt();

SAME ISSUE!

henrikekblad, 2016-01-04:
Don't use the String class in a embedded environment.

Do you get the same result if you remove everything else from the receive-function and just print the incoming data?

maxtox, 2016-01-04:
SAME ISSUE:

just left ````
void receive(const MyMessage &message)
{
  int childPin = message.sensor;
  //  Serial.println(message.getBool());
        String s = String(message.getString());
        int get_i= s.toInt();
        float get_f=s.toFloat();
  Serial.println("INT");Serial.println(get_i);
  Serial.println("FLOAT");Serial.println(get_f);
}

No other functions!
````

maxtox, 2016-01-04:
HERE LOG:
have string to Values and NORMAL getInt anf getFloat values...as Serial.println
````
`0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/13/1/0/3
STRING INT
-30704
STRING FLOAT
100368.00
INT
-30704
FLOAT
100368.00
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/58/1/0/0
STRING INT
2
STRING FLOAT
2.26
INT
2
FLOAT
2.26
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/13/1/0/3
STRING INT
-28484
STRING FLOAT
102588.00
INT
-28484
FLOAT
102588.00
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/58/1/0/0
STRING INT
0
STRING FLOAT
-0.37
INT
0
FLOAT
-0.37
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/13/1/0/3
STRING INT
-30704
STRING FLOAT
100368.00
INT
-30704
FLOAT
100368.00
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/58/1/0/0
STRING INT
2
STRING FLOAT
2.26
INT
2
FLOAT
2.26
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/13/1/0/3
STRING INT
-28484
STRING FLOAT
102588.00
INT
-28484
FLOAT
102588.00
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/58/1/0/0
STRING INT
0
STRING FLOAT
-0.37
INT
0
FLOAT
-0.37
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/13/1/0/3
STRING INT
-30704
STRING FLOAT
100368.00
INT
-30704
FLOAT
100368.00
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/58/1/0/0
STRING INT
2
STRING FLOAT
2.26
INT
2
FLOAT
2.26
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/13/1/0/3
STRING INT
-28484
STRING FLOAT
102588.00
INT
-28484
FLOAT
102588.00
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/58/1/0/0
STRING INT
0
STRING FLOAT
-0.37
INT
0
FLOAT
-0.37
````

henrikekblad, 2016-01-04:
char buf[20];
Serial.println(message.getString(buf));

maxtox, 2016-01-04:
with Output of the whole String, i see even parts from the messages send in the presentation:
"....arnings" is the rest from the ".set("for Warnings"), 0);" in the Presentation!
````
`MESSAGE as STRING
10067arningsG_FEATURE
STRING INT
10067
STRING FLOAT
10067.00
INT
10067
FLOAT
10067.00
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/58/1/0/0
MESSAGE as STRING
-0.368rningsG_FEATURE
STRING INT
0
STRING FLOAT
-0.37
INT
0
FLOAT
-0.37
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/13/1/0/3
MESSAGE as STRING
100368rningsG_FEATURE
STRING INT
-30704
STRING FLOAT
100368.00
INT
-30704
FLOAT
100368.00
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/58/1/0/0
MESSAGE as STRING
2.2588rningsG_FEATURE
STRING INT
2
STRING FLOAT
2.26
INT
2
FLOAT
2.26
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/13/1/0/3
MESSAGE as STRING
102588rningsG_FEATURE
STRING INT
-28484
STRING FLOAT
102588.00
INT
-28484
FLOAT
102588.00
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/58/1/0/0
MESSAGE as STRING
-0.368rningsG_FEATURE
STRING INT
0
STRING FLOAT
-0.37
INT
0
FLOAT
-0.37
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/13/1/0/3
MESSAGE as STRING
100368rningsG_FEATURE
STRING INT
-30704
STRING FLOAT
100368.00
INT
-30704
FLOAT
100368.00
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/58/1/0/0
MESSAGE as STRING
2.2588rningsG_FEATURE
STRING INT
2
STRING FLOAT
2.26
INT
2
FLOAT
2.26
````

maxtox, 2016-01-04:
with char buf [20];
````

`100
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/58/1/0/0
2.258
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/13/1/0/3
10
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/58/1/0/0
-0.368
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/13/1/0/3
100
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/58/1/0/0
2.258
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/13/1/0/3
10
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/58/1/0/0
-0.368
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/13/1/0/3
100
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/58/1/0/0
2.258
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/13/1/0/3
10
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/58/1/0/0
-0.368
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/13/1/0/3
100
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/58/1/0/0
2.258
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/13/1/0/3
10
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/58/1/0/0
-0.368

````

maxtox, 2016-01-04:
looks perfect!

maxtox, 2016-01-04:
so what is the issue? what can i change in the library? or my sketch?

maxtox, 2016-01-04:
if we solve it... the whole sketch as beta will be finished until end of this week...

So we could hav an MEGA as MQTT GW with attached sensors and even beginners will be in the position to maake a MULTI sensor node!!! with up to 69 (minus system pins) sensors!!!

maxtox, 2016-01-05:
Just one question... was it a mistake in my sketch? or is it a bug in the library?

Ofcourse i can do the way over the getString funktion and after convert it to Int and Float... but if there is a function there "message.getInt" why can't i (we) use it ...

Sorry for the question, but you didn't give me the last answer if it waas my mistake or it was the library Issue and i should wait for the implementation in the Dev...

Thank you in Advance!

henrikekblad, 2016-01-05:
@tekka007 and I did a quick code check yesterday but couldn't find anything wrong in the library. The payloadtype/size should be set correctly when coming in from MQTT. But we haven't done any practical tests yet so there might be something we overlooked.

maxtox, 2016-01-05:
ok... i will create the sketch with the STRING option and when you did find the bug... i will update the sketch to make it more easy :-)

Thank you!

I will wait!

maxtox, 2016-01-05:
i think i do it with 

````
void receive(const MyMessage &message)
{
  char buf[20];
  int got_int = atoi( buf );
  int got_float = atof( buf );

````

and then in the code f.e.

````
analogWrite(i, got_int*2.55);

````

i think it is the easyest way for now?!

tekka007, 2016-01-05:
Please post the entire script, I need to run some tests

closed 2016-01-05 by tekka007

maxtox, 2016-01-05:
``````
void receive(const MyMessage &message)
{
  int childPin = message.sensor;
  //  Serial.println(message.getBool());

  for (int i = 0; i<iomodus_count; i++) ////search All Sensor Variables and set them
  {
    MyMessage msg(i, iomodus[i].variableType);
    while (iomodus[i].sensorType == S_UNUSED) { i++; }  //do not waste time to check UNUSED pins

                              //********************************IF VAR1 = Temp OFFSET existst*************************************** 
    if (message.type == V_VAR1)
    {
      if ((childPin == i) && (iomodus[i].sensorType == S_TEMP) && (iomodus[i].sensorVersion == "MAX31855"))
      {
        Temp_offset[i] = message.getFloat();
      }
    }
    if (message.type == V_VAR1)
    {
      if ((childPin == i) && (iomodus[i].sensorType == S_TEMP) && (iomodus[i].sensorVersion == "NTC"))
      {
        Temp_offset[i] = message.getFloat();
      }
    }
    //********************************IF VAR2 = LPF (Low Pass Filter existst*************************************** 
    if (message.type == V_VAR2)
    {
      //***************Low Pass Filter**************
      if ((childPin == i) && (iomodus[i].sensorType == S_TEMP) && (iomodus[i].sensorVersion == "MAX31855") && (message.getFloat() <= 1.0))
      {
        ktype_LPF_rate[i] = message.getFloat();
      }
    }
    //********************************IF V_STATUS or V_LIGHT existst*************************************** 
    if (message.type == V_STATUS)
    {
      if ((childPin == i) && (iomodus[i].sensorType == S_BINARY) && (iomodus[i].sensorVersion == "RELAY"))
      {
        if (relstate[i] != message.getBool())
        {
          digitalWrite(i, message.getBool() ? RELAY_ON : RELAY_OFF);
          relstate[i] = message.getBool();
          // Store state in eeprom
          saveState(i, message.getBool());
        }
      }
    }
    //********************************IF V_PERCENTAGE or V_DIMMER existst*************************************** 
    if ((childPin == i) && (message.type == V_PERCENTAGE) && (message.getFloat() != last_int_value[i]))
    {
      Serial.println("im pwm");
      Serial.println(i);
      Serial.println(message.getFloat());
      //***************PWM Rate 0 to 100 (%)**************  
      if ((childPin == i) && (iomodus[i].sensorVersion == "PWM") && (i <= 13))
      {
        Serial.println("im digital");
        pinMode(i, OUTPUT);
        analogWrite(i, message.getFloat()*2.55); // Set Value 0-255
        last_int_value[i] = message.getFloat();
      }
      //http://www.instructables.com/id/Analog-Output-Convert-PWM-to-Voltage/
      if ((childPin == i) && (iomodus[i].sensorVersion == "PWM_5V") && (i >= 54) && (i <= 69))
      {
        Serial.println("im analog");
        pinMode(i, OUTPUT);
        analogWrite(i, message.getFloat()*2.55); // Set Value 0-255
        last_int_value[i] = message.getFloat();
      }

    }

  }
````

maxtox, 2016-01-05:
just tested with your new CPP

Looks good!!!!!!! VERY GOOD!!!

maxtox, 2016-01-05:
log
getIng 
getFloat
getBool
````
0;255;3;0;9;Sending message on topic: DOino<>MQTT/0/11/1/0/15
0;255;3;0;9;Sending message on topic: DOino<>MQTT/0/58/1/0/0
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/58/1/0/0
25
25.26
1
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/58/1/0/0
25
25.26
1
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/13/1/0/3
100
100.00
1
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/8/1/0/2
0
0.00
0
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/11/1/1/15
0;255;3;0;9;Sending message on topic: DOino<>MQTT/0/11/1/1/15
1
1.00
1
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/58/1/1/0
0;255;3;0;9;Sending message on topic: DOino<>MQTT/0/58/1/1/0
40
40.95
1
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/11/1/0/15
1
1.00
1
0;255;3;0;9;Message arrived on topic: DOino<>MQTT/0/8/1/1/2
0;255;3;0;9;Sending message on topic: DOino<>MQTT/0/8/1/1/2
0
0.00
0
0;255;3;0;9;Sending message on topic: DOino<>MQTT/0/58/1/0/0
0;255;3;0;9;Sending message on topic: DOino<>MQTT/0/58/1/0/0
````

maxtox, 2016-01-05:
I LOVE YOU ALL!!!!

commit 0ebaced5 (2016-01-05): Merge pull request #324 from tekka007/nullTerminationFix2

Null termination issue, Fixes #322
commit 39407e4a (2016-01-05): Null termination issue, Fixes #322