mysensors/MySensors #334: Error in MQTT Client Gateway Transport when use DHCP

szaffarano, 2016-01-09:
If I comment **MY_IP_ADDRESS** in MQTT Client for using dhcp I get an error:

```
In file included from lib/MySensors/MySensor.h:137:0,
from src/main.cpp:124:
lib/MySensors/core/MyGatewayTransportMQTTClient.cpp:36:24: error: 'MY_IP_GATEWAY_ADDRESS' was not declared in this scope
IPAddress _gatewayIp(MY_IP_GATEWAY_ADDRESS);
^
lib/MySensors/core/MyGatewayTransportMQTTClient.cpp:37:23: error: 'MY_IP_SUBNET_ADDRESS' was not declared in this scope
IPAddress _subnetIp(MY_IP_SUBNET_ADDRESS);
^
```

I attach a patch in the gateway that fixes the error:

```
diff --git a/libraries/MySensors/core/MyGatewayTransportMQTTClient.cpp b/libraries/MySensors/core/MyGatewayTransportMQTTClient.cpp
index d43091f..9ba820d 100644
--- a/libraries/MySensors/core/MyGatewayTransportMQTTClient.cpp
+++ b/libraries/MySensors/core/MyGatewayTransportMQTTClient.cpp
@@ -32,8 +32,10 @@ IPAddress _brokerIp(MY_CONTROLLER_IP_ADDRESS);
 
 #if defined(MY_GATEWAY_ESP8266)
        #define EthernetClient WiFiClient
-       IPAddress _gatewayIp(MY_IP_GATEWAY_ADDRESS);
-       IPAddress _subnetIp(MY_IP_SUBNET_ADDRESS);
+       #if defined(MY_IP_ADDRESS)
+               IPAddress _gatewayIp(MY_IP_GATEWAY_ADDRESS);
+               IPAddress _subnetIp(MY_IP_SUBNET_ADDRESS);
+       #endif
 #else
        byte _clientMAC[] = { MY_MAC_ADDRESS };
 #endif
```

szaffarano, 2016-01-09:
Using branch development, commit cee1dd3a4e8fd01eed999663a9292599b93a1703

henrikekblad, 2016-01-09:
Thanks for reporting. Could you please create a PR?

szaffarano, 2016-01-09:
[PR created!](https://github.com/mysensors/Arduino/pull/335)

regards

henrikekblad, 2016-02-06:
Fixed in https://github.com/mysensors/Arduino/pull/375

closed 2016-02-06 by henrikekblad