mysensors/MySensors #338: Sending message by a repeater node to itself fails

GertSanders, 2016-01-12:
A normal node can send a message addressed to itself. A repeater node can pass on messages and send messages to any node (or gateway).

The problem:
When the option MY_REPEATER_FEATURE is set, sending the same message to itself (as a repeater) fails.

The working case:
e.g. My node has address 123
When sending without MY_REPEATER_FEATURE I see:

// sending the message to myself
send: 123-123-0-0 s=1,c=1,t=47,pt=0,l=10,sg=0,st=ok:N123:alive   
// here processing of the message happens and it is decided by my sketch to pass on the received message to node N123
// passing on the stripped message to myself (since my node is nr 123).
send: 123-123-0-123 s=1,c=1,t=47,pt=0,l=5,sg=0,st=ok:alive  

When messages are received by this node, in a format Nxyz:abcdef... It will pass on all text (abcdef...) after the colon to the node with address xyz. That is what we see above.

Now for the problematic case:
If I switch in the MY_REPEATER_FEATURE I get the following:

// sending the message to myself
send: 123-123-0-0 s=1,c=1,t=47,pt=0,l=10,sg=0,st=ok:N123:alive
// message is processed by the sketch and it is again decided to send the stripped message to '123'
// sending the stripped message to myself (being '123' AND being a repeater), this time it fails.
send: 123-123-123-123 s=1,c=1,t=47,pt=0,l=5,sg=0,st=fail:alive

The question is why this happens ? I think it should not fail. This has nothing to do with radio capacitors missing or not missing.

henrikekblad, 2016-05-08:
Gert, trying to wrap my head around your issue here. 

What do expect to happen when a node sends a message to itself?

That the incomingData-method gets called with the message and nothing else?

tekka007, 2016-08-06:
@GertSanders Any updates on the use case for sending a message to itself? If not, let's close this issue.

GertSanders, 2016-08-06:
No time to investigate this in 2.0.0.

Just close it, if needed I will open een new case.

Gert

closed 2016-08-06 by tekka007