mysensors/MySensors #343: Questions

Avamander, 2016-01-13:
I've just got a few questions about the included libraries.

Wouldn't it be much more useful for everyone using MySensors if the libraries would be cloned if necessary from the originals. The thing I'm trying to say is would it be possible that you'd report bugs/improve the original libraries and then clone them to MySensors. That way the original library is developed by more people, more bugs get fixed and features added. It would benefit the end users a lot more. 

I'm mainly speaking about RF24. But there's also signing system. Wouldn't it be more beneficial if the core part of the signing system would be separately (from the MySensors) and published on github for example. That way more people could check and improve the system. 

These are just my thoughts I wanted to share. What do you think?

fallberg, 2016-01-13:
From experiences earned at work I do not favor splitting work into several gits. Opens a can of worms with dependencies and maintenance issues. I think that any code that tightly integrates with other code should be in the same repo. Signing is a typical example of this. I really do not want the signing logic to be forked out of the repo of the core lib. Especially now when I have refactored the signing implementation (as seen on development branch). Consider it a typical case where code can be designed to be separated while still be kept in the same repo, ensuring compatibility.

Avamander, 2016-01-14:
Thanks for the reply.  I really do think the signing code would be really
nice as a separate library so that it could be more easily reused.

henrikekblad, 2016-01-14:
Hm.. But it is tightly coupled with the MySensors message structure and processing. Also dependent on SHA204A emulation or hw-driver.

Not sure it would be possible to extract these parts in any meaningful way.

fallberg, 2016-01-14:
What is it supposed to be reused for? I designed it specifically for MySensors.

Avamander, 2016-01-14:
I was thinking about RF24Mesh, but it would be possible to also use it with
the entire RF24 family and other radios.

fallberg, 2016-01-14:
The signing solution is not so much intended for "radio" as it is for "communication" and it is quite protocol specific in order to be efficient. From an architectural point of view, it is novel; a nonce is exchanged which is used as salt for a hmac signature. Nothing special about it at all. So there really is not much reuse to do which is not already in its own source files. And even so, some of those source files still are MySensors specific as they operate on a MySensors specific message structure.

Avamander, 2016-01-15:
I seriously like how the signing solution handles nonces, whitelisting, hardware implementation and everything, anything else available right now comes nowhere near it. The main reason I'm writing about this is because I'd really love signing messages sent with RF24(Mesh) but I can't implement something as nice as this is.

fallberg, 2016-01-15:
Well, thanks. However, I don't have the bandwidth to maintain a security solution for several architectures and protocols. But the code is released under GPLv2 so you or anyone else is free to use it given you follow the rules of the license. If you so shoose, you can create a new git, but as I said, my code is tailored to fit the MySensors protocol. It is so for a reason; the level of security require insights into the protocol exchange. But the signing "solution" is not specific to MySensors and should not be too complicated to port. Surely someone who has developed a mesh network should be able to adapt my code to that protocol.

closed 2016-02-06 by henrikekblad