mysensors/MySensors #350: SD Card

Avamander, 2016-01-15:
I've seen EEPROM used by this library quite a bit. But would't a SD Card be a better idea, allowing more data to be stored, that data could be also modified by the PC and some things like atsha could be configured more securely by generating more random hmac for example.

tbowmo, 2016-01-15:
I'm looking at adding micro SD card support to a new GW that will be
released as official hw later this year.

henrikekblad, 2016-02-06:
Creating different drivers for storage backends is a great idea, would simplify things if/when implementing RPI support as well.

MYStorageEEPROM
MyStorageSD
MyStorageFile (RPI)
etc

tbowmo, 2016-02-06:
Will that be an extra layer on top of the SD card library? (as an example?)

henrikekblad, 2016-02-06:
Not sure. Guess we'll need things move things out from the Hw to a  MYStorageEEPROM and make a similar (thin) driver for writing/reading from SD. Just a couple of methods (which calls the SD-library) as I can see it.

tbowmo, 2016-02-07:
What if we have both eeprom and SD card?

henrikekblad, 2016-02-07:
Then you'll have to choose where you want to store the routing/node information. I can't imagine you'd want to have it stored in EEPROM and SD card at the same time?

closed 2016-11-24 by henrikekblad