mysensors/MySensors #366: ESP8266 - MQTT GW communication issue

trlafleur, 2016-01-21:
I'm running ver 1.6 from 19 Jan 2016 version with a RFM69 radio, all work well for a few hours, then it appears that the gateway stops communication with the broker, looks like the MQTT task and the WiFi task are no longer communication. Re starting the GW code fixes the issue...

I can still PING the gateway, so the WiFi stack is running......

From debug port...
```
0;255;3;0;9;Attempting MQTT connection...
0;255;3;0;9;Attempting MQTT connection...
0;255;3;0;9;read: 1-1-0 s=1,c=2,t=24,pt=0,l=0,sg=0:
0;255;3;0;9;Attempting MQTT connection...
0;255;3;0;9;Attempting MQTT connection...
0;255;3;0;9;read: 1-1-0 s=1,c=2,t=24,pt=0,l=0,sg=0:
0;255;3;0;9;Attempting MQTT connection...
0;255;3;0;9;Attempting MQTT connection...

```
I've done some more testing, It stop working in 3 to 4 hours, very consistent over the last 24 hrs. I have changes ESP8266, same issue, ESP library code is ver 2.0.. 1.6.7 and 1.6.5r5 IDE

It also appears that the code may not be trying to close, reopen the WiFi connection when there is a network error... <--- this should be implemented

Issue 2:
Also, I noted that the ESP8266 WiFi stack is still acting as an access point. Is there a way to turn off just the access point??

henrikekblad, 2016-01-21:
There is a new version of the pubsubclient (located in MySensors/drivers/pubsubclient). Maybe you could update to latest and see if stability improves?

https://github.com/knolleary/pubsubclient

henrikekblad, 2016-01-21:
Maybe related:
https://github.com/knolleary/pubsubclient/issues/95

trlafleur, 2016-01-21:
> knolleary/pubsubclient#95

I have not tested this, but it looks to be a short term fix....

As a minimum, the WiFi fix need to be in a lower layer of the code, in the
driver or transport code.
I, as a user open a connection, it should just stay open until I close it,
if for some reason, its drops , a lower layer should fix this issue
or, if it can't then give me a callback.

MQTT is a user level issue and can be solve there...

thanks

trlafleur, 2016-01-21:
looks like adding WiFi.mode (WIFI_STA); to setup turns off the Access Point in the ESP8288 radio

void setup() {
    WiFi.mode (WIFI_STA);
}

trlafleur, 2016-01-21:
This code now works fine as a test vehicle... disconnect the access point
and few time and it recovered...

will let run over the weekend and see if there are any more issues...

again, it should not be the application that fixes this, it should be the
transport layer...

Thanks for your help...

***********************************************************************************************

/**
   The MySensors Arduino library handles the wireless radio link and
protocol
   between your home built sensors/actuators and HA controller of choice.
   The sensors forms a self healing radio network with optional repeaters.
Each
   repeater and gateway builds a routing tables in EEPROM which keeps track
of the
   network topology allowing messages to be routed to nodes.

   Created by Henrik Ekblad <henrik.ekblad@mysensors.org>
   Copyright (C) 2013-2015 Sensnology AB
   Full contributor list:
https://github.com/mysensors/Arduino/graphs/contributors

   Documentation: http://www.mysensors.org
   Support Forum: http://forum.mysensors.org

   This program is free software; you can redistribute it and/or
   modify it under the terms of the GNU General Public License
   version 2 as published by the Free Software Foundation.

 *******************************

   REVISION HISTORY
   Version 1.0 - Henrik Ekblad

   DESCRIPTION
   The ESP8266 MQTT gateway sends radio network (or locally attached
sensors) data to your MQTT broker.
   The node also listens to MY_MQTT_TOPIC_PREFIX and sends out those
messages to the radio network

   LED purposes:
   - To use the feature, uncomment WITH_LEDS_BLINKING in MyConfig.h
   - RX (green) - blink fast on radio message recieved. In inclusion mode
will blink fast only on presentation recieved
   - TX (yellow) - blink fast on radio message transmitted. In inclusion
mode will blink slowly
   - ERR (red) - fast blink on error during transmission error or recieve
crc error

   See http://www.mysensors.org/build/esp8266_gateway for wiring
instructions.
   nRF24L01+  ESP8266
   VCC        VCC
   CE         GPIO4
   CSN/CS     GPIO15
   SCK        GPIO14
   MISO       GPIO12
   MOSI       GPIO13

   Not all ESP8266 modules have all pins available on their external
interface.
   This code has been tested on an ESP-12 module.
   The ESP8266 requires a certain pin configuration to download code, and
another one to run code:
   - Connect REST (reset) via 10K pullup resistor to VCC, and via switch to
GND ('reset switch')
   - Connect GPIO15 via 10K pulldown resistor to GND
   - Connect CH_PD via 10K resistor to VCC
   - Connect GPIO2 via 10K resistor to VCC
   - Connect GPIO0 via 10K resistor to VCC, and via switch to GND
('bootload switch')

    Inclusion mode button:
   - Connect GPIO5 via switch to GND ('inclusion switch')

   Hardware SHA204 signing is currently not supported!

   Make sure to fill in your ssid and WiFi password below for ssid & pass.
*/

//#define MY_RFM69HW true
#define MY_RFM69_FREQUENCY     RF69_915MHZ

#include <EEPROM.h>
#include <SPI.h>

// Enable debug prints to serial monitor
#define MY_DEBUG

// Use a bit lower baudrate for serial prints on ESP8266 than default in
MyConfig.h
#define MY_BAUD_RATE 9600

// Enables and select radio type (if attached)
//#define MY_RADIO_NRF24
#define MY_RADIO_RFM69

#define MY_GATEWAY_MQTT_CLIENT

// Set this nodes subscripe and publish topic prefix
#define MY_MQTT_PUBLISH_TOPIC_PREFIX "mygateway1-out"
#define MY_MQTT_SUBSCRIBE_TOPIC_PREFIX "mygateway1-in"

// Set MQTT client id
#define MY_MQTT_CLIENT_ID "mysensors-1"

// Enable these if your MQTT broker requires usenrame/password
//#define MY_MQTT_USER "username"
//#define MY_MQTT_PASSWORD "password"

// Set WIFI SSID and password
#define MY_ESP8266_SSID "xxxxxx"
#define MY_ESP8266_PASSWORD "sxxxxxxx"

// Enable MY_IP_ADDRESS here if you want a static ip address (no DHCP)
#define MY_IP_ADDRESS 192,168,xx,xx

// If using static ip you need to define Gateway and Subnet address as well
#define MY_IP_GATEWAY_ADDRESS 192,168,xx,xx
#define MY_IP_SUBNET_ADDRESS 255,255,255,0

// MQTT broker ip address.
#define MY_CONTROLLER_IP_ADDRESS 192,xx,xx,xx

// The MQTT broker port to to open
#define MY_PORT 1883

/*
  // Flash leds on rx/tx/err
  #define MY_LEDS_BLINKING_FEATURE
  // Set blinking period
  #define MY_DEFAULT_LED_BLINK_PERIOD 300

  // Enable inclusion mode
  #define MY_INCLUSION_MODE_FEATURE
  // Enable Inclusion mode button on gateway
  #define MY_INCLUSION_BUTTON_FEATURE
  // Set inclusion mode duration (in seconds)
  #define MY_INCLUSION_MODE_DURATION 60
  // Digital pin used for inclusion mode button
  #define MY_INCLUSION_MODE_BUTTON_PIN  3

  #define MY_DEFAULT_ERR_LED_PIN 16  // Error led pin
  #define MY_DEFAULT_RX_LED_PIN  16  // Receive led pin
  #define MY_DEFAULT_TX_LED_PIN  16  // the PCB, on board LED
*/

#include <ESP8266WiFi.h>
#include <MySensor.h>

unsigned long last_wifi_reconnect_attempt = 0;
unsigned long reconnect_attempt = 0;

void wifiConnect() {
  WiFi.hostname("esp8266");
  WiFi.mode(WIFI_STA);
  WiFi.begin(MY_ESP8266_SSID, MY_ESP8266_PASSWORD);
}

void setup() {
    WiFi.mode (WIFI_STA);
}

void presentation() {
  // Present locally attached sensors here
}

void loop() {

  if (WiFi.status() != WL_CONNECTED) {
    unsigned long now = millis();

    if (now - last_wifi_reconnect_attempt > 20000UL ||
last_wifi_reconnect_attempt == 0) {
      Serial.print("Attempting to connect to WiFi: ");
      Serial.print(reconnect_attempt);
      Serial.println("\r\n");
      last_wifi_reconnect_attempt = now;
      wifiConnect();
      reconnect_attempt++;    // adjust re attemp count
    }
    return;
  }

}

trlafleur, 2016-01-22:
It appears that this is NOT the issues, looks like the WiFi and MQTT task
are disconnected...
WiFi is still connected to AP...

will looks for another solution...

closed 2016-02-06 by fallberg