mysensors/MySensors #368: GatewaySerial not receiving data with MY_WITH_LEDS_BLINKING_INVERSE

Kolbi, 2016-01-22:
I just tried to compile a GatewaySerial (https://github.com/mysensors/Arduino/blob/development/libraries/MySensors/examples/GatewaySerial/GatewaySerial.ino) with MY_WITH_LEDS_BLINKING_INVERSE enabled. It compiles but the Gateway can't receive any data, uncommenting this line and recompiling solve this issues. So it must be this function.
Thank you.

mrred2k, 2016-02-14:
I just wanted to contribute here: I tried changing the pins for the 3 LEDS but the pinchange did not have any effect. Did not expierenced any problems compiling thought.

tekka007, 2016-08-06:
@Kolbi Is this still an issue with 2.0.0?

Kolbi, 2016-08-08:
@tekka007 good question :) haven't tested it since a while.

sincze, 2016-08-08:
Mmm, I seem to suffer from a similar issue not with MY_WITH_LEDS_BLINKING_INVERSE but with MY_WITH_LEDS_BLINKING. The Serial Gateway 2.0.0. seems to sleep. Only after several triggers it awakes and the leds start flashing when i trigger a doorsensor.. Leaving serial gateway without activity for a while same results. I have to try to see if i can open 2 serial monitors that will monitor the serial gateway and the other one the sensor to see what exactly the logs will say.

henrikekblad, 2016-11-24:
Fixed #640

closed 2016-11-24 by henrikekblad