mysensors/MySensors #369: W5100 Gateway / Ethernet Gateway does not compile`/ BUG in MyGatewayTransportEthernet.cpp?

maxtox, 2016-01-23:
Hi,

i am using the latest Dev Version of libraries:

if i just take the Example "Gatewayw5100" no changes:
```
D:\03_Daten\07_Arduino\01_Projekte\libraries\MySensors\examples\GatewayW5100\GatewayW5100.ino:113:22: fatal error: Ethernet.h: No such file or directory

 #include <Ethernet.h>

compilation terminated.
````

even it is an Ethernet library in D:\03_Daten\07_Arduino\ORIGArduino\hardware\MySensors\samd\libraries\Ethernet\src as included in ZIP file!

if i copy the library from HARDWARE folder to original Library folder, i get this error:

````
D:\03_Daten\07_Arduino\01_Projekte\libraries\Ethernet\src\Ethernet.cpp: In member function 'int EthernetClass::begin(uint8_t*, long unsigned int, long unsigned int)':

D:\03_Daten\07_Arduino\01_Projekte\libraries\Ethernet\src\Ethernet.cpp:19:3: error: 'SPI_ETH' was not declared in this scope

   SPI_ETH.beginTransaction(SPI_ETHERNET_SETTINGS);

   ^

D:\03_Daten\07_Arduino\01_Projekte\libraries\Ethernet\src\Ethernet.cpp: In member function 'void EthernetClass::begin(uint8_t*, IPAddress, IPAddress, IPAddress, IPAddress)':

D:\03_Daten\07_Arduino\01_Projekte\libraries\Ethernet\src\Ethernet.cpp:68:3: error: 'SPI_ETH' was not declared in this scope

   SPI_ETH.beginTransaction(SPI_ETHERNET_SETTINGS);

   ^

D:\03_Daten\07_Arduino\01_Projekte\libraries\Ethernet\src\Ethernet.cpp: In member function 'int EthernetClass::maintain()':

D:\03_Daten\07_Arduino\01_Projekte\libraries\Ethernet\src\Ethernet.cpp:89:9: error: 'SPI_ETH' was not declared in this scope

         SPI_ETH.beginTransaction(SPI_ETHERNET_SETTINGS);

         ^

D:\03_Daten\07_Arduino\01_Projekte\libraries\Ethernet\src\Ethernet.cpp: In member function 'IPAddress EthernetClass::localIP()':

D:\03_Daten\07_Arduino\01_Projekte\libraries\Ethernet\src\Ethernet.cpp:107:3: error: 'SPI_ETH' was not declared in this scope

   SPI_ETH.beginTransaction(SPI_ETHERNET_SETTINGS);

   ^

D:\03_Daten\07_Arduino\01_Projekte\libraries\Ethernet\src\Ethernet.cpp: In member function 'IPAddress EthernetClass::subnetMask()':

D:\03_Daten\07_Arduino\01_Projekte\libraries\Ethernet\src\Ethernet.cpp:116:3: error: 'SPI_ETH' was not declared in this scope

   SPI_ETH.beginTransaction(SPI_ETHERNET_SETTINGS);

   ^

D:\03_Daten\07_Arduino\01_Projekte\libraries\Ethernet\src\Ethernet.cpp: In member function 'IPAddress EthernetClass::gatewayIP()':

D:\03_Daten\07_Arduino\01_Projekte\libraries\Ethernet\src\Ethernet.cpp:125:3: error: 'SPI_ETH' was not declared in this scope

   SPI_ETH.beginTransaction(SPI_ETHERNET_SETTINGS);

   ^
````

if i renew the Ethernet library with the one from Arduino github (1.0.5)

i can compile!!!! NO Errors!

but i want to use UDP to send messages to my controler!
so i  TURN ON 
````
// Enable to UDP          
#define MY_USE_UDP
````
and try to compile , i get this  ERROR report:

````
In file included from D:\03_Daten\07_Arduino\01_Projekte\libraries\MySensors/MySensor.h:221:0,

                 from D:\TEMP\arduino_a6199089dc15f417b80dd389b859fb7d\GatewayW5100.ino:114:

D:\03_Daten\07_Arduino\01_Projekte\libraries\MySensors/core/MyGatewayTransportEthernet.cpp: In function 'bool gatewayTransportAvailable()':

D:\03_Daten\07_Arduino\01_Projekte\libraries\MySensors/core/MyGatewayTransportEthernet.cpp:263:36: error: no match for 'operator[]' (operand types are 'inputBuffer' and 'int')

    _ethernetServer.read(inputString[0].string, MY_GATEWAY_MAX_RECEIVE_LENGTH);

                                    ^

In file included from D:\03_Daten\07_Arduino\01_Projekte\libraries\MySensors/MySensor.h:28:0,

                 from D:\TEMP\arduino_a6199089dc15f417b80dd389b859fb7d\GatewayW5100.ino:114:

D:\03_Daten\07_Arduino\01_Projekte\libraries\MySensors/core/MyGatewayTransportEthernet.cpp:265:56: error: no match for 'operator[]' (operand types are 'inputBuffer' and 'int')

    debug(PSTR("UDP packet received: %s\n"), inputString[0].string);

                                                        ^

D:\03_Daten\07_Arduino\01_Projekte\libraries\MySensors/core/MySensorCore.h:36:40: note: in definition of macro 'debug'

 #define debug(x,...) hwDebugPrint(x, ##__VA_ARGS__)

                                        ^

In file included from D:\03_Daten\07_Arduino\01_Projekte\libraries\MySensors/MySensor.h:221:0,

                 from D:\TEMP\arduino_a6199089dc15f417b80dd389b859fb7d\GatewayW5100.ino:114:

D:\03_Daten\07_Arduino\01_Projekte\libraries\MySensors/core/MyGatewayTransportEthernet.cpp:266:50: error: no match for 'operator[]' (operand types are 'inputBuffer' and 'int')

    return protocolParse(_ethernetMsg, inputString[0].string);
````

tbowmo, 2016-01-23:
What platform are you compiling for, and what environment are you using? (arduino version etc)

There should be an ethernet library in your standard arduino libraries, which is used by the standard arduino platform (AVR hardware)

The ethernet lib that you reffer to, is only meant to be used by the new SAMD GW, and have been modified to be able to run on this hardware platform (we are using more than one SPI port on the GW, and the standard ethernet library is hardwired to use a specific SPI port)

So it seems that your arduino install might be broken?

maxtox, 2016-01-23:
I am compiling for "ARDUINO Mega 2560"

TCP is running with the Etherent library 1.5.0 (Standard to download from library manager of arduino"

but

UDP is not working, does not compile...

I am using Arduino IDE 1.6.7

tbowmo, 2016-01-23:
I'm not familiar with that hardware platform..  As I'm more focused towards SAMD architecture at the moment :)

maxtox, 2016-01-23:
welcome to Arduino ;-)

tbowmo, 2016-01-23:
I mean that I'm not familiar with the mega 2560 hardware platform :) I'm using standard 328p.. and the last 6 months SAMD arch as well, adding support for the SAMD in arduino and mysensors

maxtox, 2016-01-23:
oh...COOL!!! Thank you " Newer used SAMD

maxtox, 2016-02-01:
still no idea how to solve it?

tbowmo, 2016-02-02:
you mention that you have taken the ethernetlib from arduino 1.0.5, which is rather old.

Have you tried with the latest from arduino 1.6.7? (Should be in the download when you install that version of the IDE)

maxtox, 2016-02-02:
yes...tested ALL versions...1.6.7 too

closed 2016-02-06 by fallberg