mysensors/MySensors #371: New Sensor type

trlafleur, 2016-01-24:
Please consider adding: S_PRESSURE, this could be used for water pressure, air pressure (like in a compressor tank) or ??

Current S_BARO  imply Barometric pressure, not absolute or reference pressure like from a gauge...
One could #define S_PRESSURE S_BARO, but this is confusing..
and S_PRESSURE is generic, S_BARO is specific

my two bits...

trlafleur, 2016-02-09:
another useful sensor type would be S_RELAY

my two bits...

henrikekblad, 2016-02-09:
What differs relay from S_BINARY(/S_LIGHT)?

trlafleur, 2016-02-09:
One could use almost any sensor type for a relay (like: power, door, binary, lock, custom), but having a S_RELAY and/or S_SWITCH is an explicit type.... and not generic...

my two bits...

trlafleur, 2016-02-27:
One of the problems, is that over time, the Presentation Message's have gotten mixed with generic type and specific type...

alarm, weather, Pressure, light, sprinkler, HVAC, and other are generic and could have subtypes..
but
door, motion and smoke, would make more sense if they were under Alarm, same for
light_level, RGB(w)_light, light and others...

we appear to be just adding on for compatibility, and not doing a good job at layering protocol...

just my two bits...

It all works, but maybe it time for the next major release to re look at this...)