mysensors/MySensors #379: Move EEPROM_LOCAL_CONFIG_ADDRESS to end of EEPROM

henrikekblad, 2016-01-30:
We should move the user/local config are to end end of EEPROM so we safely can add new internal configurations without affecting/shifting the local config.

d00616, 2017-05-02:
If the "end of EEPROM" is the highest available cell address, there is a problem with the NVRAM class. The highest written cell determines the possible change log size.

A smaller change log stands for more page erase cycles. This means to shorten the lifetime of the internal flash memory. I don't know how often the EEPROM is written. In this situation 10,000,000 writes are available.

The NVRAM emulates 3072 bytes. It could be an option move the EEPROM_LOCAL_CONFIG_ADDRESS to 1k.

mfalkvidd, 2018-05-20:
Maybe saveState could take care of changes to EEPROM_LOCAL_CONFIG_ADDRESS internally instead?

flatsiedatsie, 2019-02-25:
I am running into this issue with the [433 signal cloner](https://forum.mysensors.org/topic/10094/candle-433-hub-a-universal-433mhz-signal-cloner). It stores a lot of signal data in eeprom. But because I want it to be safe for users to upgrade to a new version of MySensors without losing their recordings, I am now starting the recordings at the halfway mark of the eeprom, at position 512.

With the MY_SIMPLE_SECURITY option the eeprom claims to be empty from position 413 onwards. So 99 bytes will stay unused. That's "99 problems, but it's not big ones".

Still, if there was a way of grabbing it, it would be nice.

mfalkvidd, 2019-02-25:
For your use case, why can't you start right at the end of the security stuff?