mysensors/MySensors #382: RFM69 Pins

trlafleur, 2016-02-02:
The code below will NOT override the pins defined in RFM69.h
I'm testing in ESP8266MQTTClient

I have tried prior and after #include <MySensor.h>

Only works if I make changes in RFM69.h file

we need to be able to change pin numbers in application code.

~~~
// If using OSH board
#define RF69_SPI_CS  2     // D2
#define RF69_IRQ_PIN 15    // D8

#include <ESP8266WiFi.h>
#include <MySensor.h>

// If using OSH board
#define RF69_SPI_CS  2     // D2
#define RF69_IRQ_PIN 15    // D8

~~~

trlafleur, 2016-02-05:
I made these changes to RFM69.h to use various boards with this radio.. (ESP8266)
It should be setup using #ifndef for each pin... with visibility to the .ino application file... but that's for another day...
~~~
#ifndef RFM69_h
#define RFM69_h
#include <Arduino.h>            // assumes Arduino IDE v1.0 or greater

#define RF69_MAX_DATA_LEN      61 // to take advantage of the built in AES/CRC we want to limit the frame size to the internal FIFO size (66 bytes - 3 bytes overhead)

//#endif

// INT0 on AVRs should be connected to RFM69's DIO0 (ex on ATmega328 it's D2, on ATmega644/1284 it's D2)
#if defined(__AVR_ATmega168__) || defined(__AVR_ATmega328P__) || defined(__AVR_ATmega88) || defined(__AVR_ATmega8__) || defined(__AVR_ATmega88__)
  #define RF69_IRQ_PIN          2
  #define RF69_IRQ_NUM          0
  #define RF69_SPI_CS           SS // SS is the SPI slave select pin, for instance D10 on atmega328
//#endif
#elif defined(__AVR_ATmega644P__) || defined(__AVR_ATmega1284P__) || defined(__AVR_ATmega2560__)
  #define RF69_IRQ_PIN          2
  #define RF69_IRQ_NUM          2
  #define RF69_SPI_CS           SS  // SS is the SPI slave select pin, for instance D10 on atmega328
//#endif
#elif defined(__AVR_ATmega32U4__)
  #define RF69_IRQ_PIN          3
  #define RF69_IRQ_NUM          0
  #define RF69_SPI_CS           SS  // SS is the SPI slave select pin, for instance D10 on atmega328
//#endif
#elif defined(ARDUINO_ARCH_ESP8266)  
  // ESP8266 should fall into this category now
  // TODO !! Untested code! Entering unknown territory...
  #define RF69_IRQ_PIN          15
  #define RF69_IRQ_NUM          0
  #define RF69_SPI_CS           2	// SS is the SPI slave select pin, for instance D10 on atmega328, 6 for OSH board ESP8266
#endif
~~~

henrikekblad, 2016-02-06:
The default pin assignment should take place in MyConfig.h.

If you decide to provide a PR you should move this this code over to MyConfig.h and make it overridable from sketch by using #ifndef.

mfalkvidd, 2017-05-22:
This is now possible by using the following defines:
MY_RF69_SPI_CS
MY_RF69_IRQ_PIN

closed 2017-05-22 by mfalkvidd