mysensors/MySensors #383: RFM69 Ethernet Gateway

trlafleur, 2016-02-03:
Ethernet Gateway and RFM69.c code need to be upgraded to support softSPI for this radio

Thanks

henrikekblad, 2016-02-06:
If you can, please provide a PR for this one.

closed 2017-01-04 by trlafleur

gieemek, 2017-07-23:
I modified RFM69 driver and MyConfig.h file in MySensors 2.1.1 library to work by SoftSPI bus:
https://github.com/gieemek/RFM69_softSPI-for-MySensors-2.1.1

Feel free to implement it to your fantastic library if you want.

trlafleur, 2017-07-23:
FYI:    Newer 5100 Ethernet cards have the chip select issued fixed on the board allowing it to properly share an SPI bus with the rfm69 radio.   

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

https://github.com/gieemek/RFM69_softSPI-for-MySensors-2.1.1

gieemek, 2017-07-23:
Of course, 5100 ethernet cards don't have pins for SPI bus - this card uses SPI bus, so you can't join another module to this bus.
But you can create software SPI bus on your Arduino, which can use any pins. SoftSPI library is used by MySensors platform to use W5100 with NRF24L01 - the same is possible with RFM69.
I modified RFM69 driver and MyConfig.h file to allow SoftSPI with RFM69 radio module.