mysensors/MySensors #389: MQTTGateway stops publishing

ferluszt, 2016-02-07:
I use the MQTT Gateway with sketch from [mysensors.org](http://www.mysensors.org/build/mqtt_gateway.)
After a while the gateway stops publishing mqtt messages.
```
MyMQTT/20/0/V_TEMP
>>30 18 00 12 4D 79 4D 51 54 54 2F 32 30 2F 30 2F 56 5F 54 45 4D 50 31 37 2E 30 
0;0;3;0;9;read: 22-22-0 s=0,c=1,t=0,pt=7,l=5,sg=0:5.0
MyMQTT/22/0/V_TEMP
>>30 17 00 12 4D 79 4D 51 54 54 2F 32 32 2F 30 2F 56 5F 54 45 4D 50 35 2E 30 
0;0;3;0;9;read: 20-20-0 s=1,c=1,t=13,pt=2,l=2,sg=0:5
MyMQTT/20/1/V_DISTANCE
>>30 19 00 16 4D 79 4D 51 54 54 2F 32 30 2F 31 2F 56 5F 44 49 53 54 41 4E 43 45 35 
0;0;3;0;9;read: 20-20-0 s=0,c=1,t=0,pt=7,l=5,sg=0:17.0
MyMQTT/20/0/V_TEMP
>>30 18 00 12 4D 79 4D 51 54 54 2F 32 30 2F 30 2F 56 5F 54 45 4D 50 31 37 2E 30 
0;0;3;0;9;read: 20-20-0 s=1,c=1,t=13,pt=2,l=2,sg=0:14
MyMQTT/20/1/V_DISTANCE
>>30 1A 00 16 4D 79 4D 51 54 54 2F 32 30 2F 31 2F 56 5F 44 49 53 54 41 4E 43 45 31 34 
0;0;3;0;9;read: 20-20-0 s=0,c=1,t=0,pt=7,l=5,sg=0:17.0
MyMQTT/20/0/V_TEMP
>>30 18 00 12 4D 79 4D 51 54 54 2F 32 30 2F 30 2F 56 5F 54 45 4D 50 31 37 2E 30 
0;0;3;0;9;read: 20-20-0 s=1,c=1,t=13,pt=2,l=2,sg=0:4
MyMQTT/20/1/V_DISTANCE
>>30 19 00 16 4D 79 4D 51 54 54 2F 32 30 2F 31 2F 56 5F 44 49 53 54 41 4E 43 45 34 
<<C0 00 
>>D0 00 
0;0;3;0;9;read: 20-20-0 s=0,c=1,t=0,pt=7,l=5,sg=0:17.0
MyMQTT/20/0/V_TEMP
>>30 18 00 12 4D 79 4D 51 54 54 2F 32 30 2F 30 2F 56 5F 54 45 4D 50 31 37 2E 30 
0;0;3;0;9;read: 20-20-0 s=1,c=1,t=13,pt=2,l=2,sg=0:4
MyMQTT/20/1/V_DISTANCE
>>30 19 00 16 4D 79 4D 51 54 54 2F 32 30 2F 31 2F 56 5F 44 49 53 54 41 4E 43 45 34 
0;0;3;0;9;read: 20-20-0 s=0,c=1,t=0,pt=7,l=5,sg=0:17.0
MyMQTT/20/0/V_TEMP
>>30 18 00 12 4D 79 4D 51 54 54 2F 32 30 2F 30 2F 56 5F 54 45 4D 50 31 37 2E 30 
0;0;3;0;9;read and forward: 21-149-20 s=85,c=3,t=85,pt=7,l=13,sg=0:24
MyMQTT/20/0/V_TEMP17.044
>>30 35 00 2D 4D 79 4D 51 54 54 2F 32 30 2F 30 2F 56 5F 54 45 4D 50 31 37 2E 30 34 34 00 00 00 00 2F 32 30 2F\00\00
```

henrikekblad, 2016-02-07:
Most probably a powering issue.

We won't fix any bugs in the 1.5/MQTTGateway. You should try the client gateway in dev instead.

closed 2016-02-07 by henrikekblad

ferluszt, 2016-02-07:
thanks for your comment. I've got the client gateway working from dev branch. Is there any further information about the dev branch, because I had to adjust the openhab items config due to the changing topic structure?

henrikekblad, 2016-02-07:
http://forum.mysensors.org/topic/2352/guide-setting-up-and-testing-mqtt-client-gateway