mysensors/MySensors #393: Problem with MySensor.h including .cpp files

TD22057, 2016-02-11:
I ran into compilation errors in a sketch that are due to the way that MySensor.h is written in the development branch.  That file includes a lot of .cpp code files in #include directives.  That means that every #define macro, every function, and every variable defined in those files can conflict with my own code in the sketch and potentially other libraries that I include (not sure about that one).

I hit this first with a variable.  I was using a variable (_address) which is defined in the transport cpp files.  I figured this was a mistake so I added 'static' to the front of the _address variable in the transport files since that variable is only needed by the transport functions and should be private (all variables should be limited to the minimum scope they need to be used in - especially in libraries).  But that fix didn't work - because the radio cpp file is being included into my main sketch code via MySensor.h.

The I hit this problem again when I defined a variable "TIME_OUT" which happens to conflict with a macro definition in the RFM69.cpp file.  Again - same problem.  By including the cpp file, every single definition is being added to the same scope as the sketch.  Given the number of files being included, this is a large number of variables and macro which can have common names (f2, TIME_OUT are just a couple I found).  Given that all of these definitions are basically hidden in cpp files (which should normally be the private implementation part of a library), I don't think this working very well.  

It's not obvious to me why including cpp files in MySensor.h (or anywhere) is a good idea (I'm sure there is a reason for it of course).  IMHO, it would be a lot better to keep cpp files as separate compilation units and limit all names to the minimum scope they need for use (adding static for most, if not all variables).  Even better would be to be eliminate macros in headers whenever possible and use namespaces for all functions and global variables to avoid all name collisions.

I'm willing to help out with this work if you agree that it's a good idea.  Just let me know...

henrikekblad, 2016-02-11:
The reason for looking like this is how the Arduino IDE works and how it links libraries based on the includes in the sketch. 

The goal was to allow you to enable features and control settings from the sketch. I.e SOFT_SPI, different radio backends etc.  This is because we're using codebender and many people is posting their examples on the forum etc. So all the config needs to be included in sketch, if possible. 

The only possible way of doing this is by including the cpp files directly based on defines from sketch. 
At least prior to IDE verion 1.5.7 which have introduced news in this respect. I haven't really played with the new features. You're welcome to see if you can find a better way of doing things in the new IDE.

There's no excuse for the none static local variables. Would be nice if someone could go through and limit their scope. Be aware that some variables I.e. _msg is actually used between files. 

RFM69.cpp hasn't been adopted at all. It will probably get refactored into something smaller like the current RF24 PR by @tekka007 
.

TD22057, 2016-02-11:
Adding static won't help - since the files get combined into a single unit, static just limits them to that unit (which is the users sketch).  

To restate what I think you're saying:  The goal is to have the user control code getting added/used from in their own sketch via #define variables.  This means that the library code itself has to see those variables which it can't do if it's a traditional library where the library code would be compiled by itself and then linked into the users code.  (FYI - here is the [IDE library spec](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification))

I'm going to start thinking out loud below about ways to reorganize the system to avoid this.  But if we want a "simpler" solution, it might be good enough to adopt a set of coding guidelines.  I think part of the problem is that people writing the code may not realize that every cpp file gets included in the users sketch instead of being a private, implementation unit.  So to avoid problems, coding/style guidelines could dictate that all symbols (variables, static functions, local class/struct defintions) be put in MyPrivate namespace or prefixed with MY_PRIV and all #defines have a prefix like MY_PRIV_[FLAG].  Even better would be a cpp file specific prefix (MY_RFM69_[FLAG]). This would virtually eliminate the possibility of conflicting with other libraries and any user code in the sketch.

I think the biggest use of the #defines is in controlling the library behavior.  "Controlling" in this case means things like pin numbers, on/off flags, and general behavior.  This is normally stuff that I would consider as constructor arguments to a class (or to a begin/init function in the setup).  For example, the radio gets constructed with the pin numbers it's wired to and a flag to turn encryption on or off.  So in a traditional(?) coding scenario, the user would just build these classes with the information and everything would work fine.   

It seems like one of the goals of the dev branch is to eliminate the user from having to do this and to control everything via #defines.  My personal opinion is that this is unnecessary and makes it harder to understand and use the system but given the kinds of posts I see on the forum, I can see why this might be a useful goal.

Without having done any work on it, my guess is that this is a solvable problem.  The main part of the MySensors auto-generated code is the radio and message processing.  That "code" could still be included in the user's sketch the way it is now and the #defines would be passed as constructor or begin() function arguments to control behavior.  This can also control things like which radio gets used.  So my idea is might be stated as saying "design all the classes normally (fully encapsulated, good OO practices) including abstract interfaces for components, have them communicate with each other via handles to instance variables of other classes, and have the #defines control which classes get instantiated and how they're constructed".

Something like (not real obviously):

```
// MY_Rfm69Transport and MY_NrfTransport inherit from a common API base class

#ifdef MY_RADIO_RFM69
#  include "MyTransportRFM69.hpp"
   MY_Rfm69Transport MY_transport( FLAG1, FLAG2, ...);
#else 
#  ifdef MY_RADIO_NRF
#     include "MyTransportRFM69.hpp"
      MY_NrfTransport MY_transport( FLAG1, FLAG2, ...);
#  endif
#endif

// pass MY_transport variable to another class constructor here to use it
```

So instead of including cpp files for all the code and classes, it just needs to include a cpp file to 
build the correct classes with the requested flags.  Which code gets used is controlled by which classes get build and passed to other classes which would hold handles to them.  This might mean moving back to the more class based designs of v1.5 vs the current free functions that exist on the dev branch.  The IDE library manager will still compile the whole library, but the linker will eliminate all the classes and code that's not used so there is no penalty for having all the different classes it in the library.  

This kind of solution probably leads to slightly larger sketch sizes since it's harder to remove large blocks of code of code from inside the library.  That can still be mostly accomplished through API's classes.  For example - build a dummy LedBlinking API which has methods for blinking the LED's.  Create a DummyLed version which doesn't nothing (minimum code size) and a full version which does the blinking and pass it to the class which needs to call it.  This kind of API design might be useful for other reasons as well since it would allow people to easily override behavior when they want to.

I'm not opposed to giving this kind of design a try - but I'd prefer not to bother unless the main developers think it's a good idea.   It's probably a lot of work to try out (especially for an outsider like myself) so I'm curious what everyone else thinks of it...

fallberg, 2016-02-11:
This sounds uncomfortably a lot like C++ to me. We do not want more C++ misery in the core.

TD22057, 2016-02-11:
OK - interesting opinion.  I would have said it's organization, not misery.  C++ can be abused like any language but when done properly (which isn't that difficult), it's makes things a lot easier to understand and maintain.  

I still think having all the cpp code included together is going to cause serious maintenance headaches and problems - especially because code is written as if each file is a stand alone unit (which is what a c/cpp file normally is).  Given the use of #defines and local/global variables (and the fact that it's never clear in the code when a variable is meant to be local or global), including all of them together is going to cause conflicts (like it did with me) sooner or later.  People write each code file as if it's an atomic unit - which it isn't in this case.  Each code file needs to get written knowing that they're all going to be merged before compiling which means naming of macros and symbols is important.

Do you have any thoughts on other ways to fix this?  What do you think of the naming requirement?

fallberg, 2016-02-11:
Well, we saved literally K:s of flash by removing C++ stuff form the library without sacrificing functionality. The c++ compiler Arduino uses is miserable at code optimization.
You can implement the same standalone unit approach using C. Just expose a number of function pointers to predefined behaviors and let the caller feed them back when initializing the library and you can customize it's behavior and let the linker throw away unused code.

TD22057, 2016-02-11:
OK - fine.  But what do you think about how to solve the naming conflicts of including all the cpp code together?

fallberg, 2016-02-11:
I don't like it either, and Henrik gave me the same explanation when I ranted on including cpp files in headers. But I dislike the arduino IDE and it's twisted take on a build system more. I think the most immediate action to take is to prefix all global symbols used by the core library with some less likely value like _MYS_ or similar. How to handle customization of a pre-built library can be solved later on.

henrikekblad, 2016-10-01:
Naming conflicts and scope of variables has improved since this issue.

closed 2016-10-01 by henrikekblad