mysensors/MySensors #398: W5100 Ethernet gateway does not support multiple connections

soif, 2016-02-13:
Following our [discussion on MySensors forums](http://forum.mysensors.org/topic/3114/other-controller-get-disconnected-when-sending-any-command-to-the-eth-gateway-v2-0/5), a W5100 based ethernet Gateway does not allow multiple connections, because it is not [currently coded](https://github.com/mysensors/Arduino/blob/7e3f5d2d5410117367dbb52b5c1e578f40b2e590/libraries/MySensors/core/MyGatewayTransportEthernet.cpp#L306-L330)

I've done some google and found an [interesting blog post](http://subethasoftware.com/2013/04/08/arduino-ethernet-and-multiple-socket-server-connections/) about supporting multiple telnet connections (Up to 4 I believe) on the W5100. My C++ skills are *very limited*, but It seems that one of the poster (in the comments), submited an example of code to make this possible. 

**If there is a C++ hero around there :+1:  willing to contribute to implement it**, the following code may be (or not?) a good start:

Here it the copy/pasted code from the blog post linked above : 
````c++
/* exampleMultiTelnetServer.ino */
/* 
 *  Created: May 21, 2015
 *
 *  Created by Gene Reeves to demonstrate how the W5100 eithenet handles multiple connections
 *  using a single server object instance.
 * 
 *  once the sketch is loaded on your arduino and you see the start up banner display in the
 *  Serial Monitor, fire up your favorite telnet client and connect a few instances.
 *  
 *  from the serial monitor you can send text to one or all clients. Use '*" for the ID to send
 *  to all, otherwise use the ID you wish to send to, using this format to send,
 *   "ID:TEDXT_TO_SEND".  That should make everything as clear as mud  :-)
 *
 *  Dependencies:
 *
 *  uses LinkedList class from here https://github.com/ivanseidel/LinkedList
 *  to store list of connected clients.
 *
 *  uses StringStream class from here https://gist.github.com/cmaglie/5883185
 *  Not sure why this was not a part of the StringObject to start with..
 *
 *  uses elapsedMillis class from here http://playground.arduino.cc/Code/ElapsedMillis
 *  much improved version (for Teensy brand mcu's) here https://www.pjrc.com/teensy/td_timing_elaspedMillis.html
 *  Unless I am mistaken, I beleive this code was originated by Paul Stoffregen of www.pjrc.com
 *  If you are not famialar with Paul or his Teensy microcontroller, you would do yourself a 
 *  solid to spend some time reviewing his code and you can't beat the Teensy 3.1 for 
 *  price/performance/support!!!   
 *  Teensy 3.1 - 32bit-ARM  mcu @ 72MHz (overclockable to 168MHz) w/ 256K flash / 64K ram / USB
 *  for under $20.  https://www.pjrc.com/teensy/index.html
 *
 *  BTW, I am in no way affilaited with Paul or PJRC, I just think he built an awesome mcu.
 *
 */
 
 
#include <SPI.h>
#include <LinkedList.h>
#include <elapsedMillis.h>
#include <StringStream.h>
#include <Arduino.h>
#include <EthernetUdp.h>
#include <EthernetServer.h>
#include <EthernetClient.h>
#include <Ethernet.h>
#include <Dns.h>
#include <Dhcp.h>
#include <IPAddress.h>
#include <Server.h>
#include <Client.h>
 
 
/*  DEFINES  */
                      
/*****************************************************/
/*  Change the following to suite your environment.  */
#define TELNET_SERVER_PORT 23
#define TELNET_SERVER_MAC 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED
#define TELNET_SERVER_IP 192, 168, 1, 177
#define TELNET_SERVER_GATEWAY 192, 168, 1, 1
#define TELNET_SERVER_NETMASK 255, 255, 255, 0
#define SERIAL_BAUD 115200
[... 451 lines cut ...]
  Serial.print(F(" was connected for "));
  Serial.print(cli->elapsed() / 1000);
  Serial.print(F(" seconds, and has disconnected."));
  Serial.flush();
 
  // remove from list of clients
  ClientList.drop(cli->index);
  (*cli) = (*(ClientItem *)(unsigned long)0);
  // print connected count
  console_print_connected_count();
 
  // this causes seg fault ??
  //delete cli;
}
 
size_t send_to(int idx, const String &str)
{
  return ClientList.send(str, idx);
}
 
size_t send_to_all(const String &str)
{
  return ClientList.send2All(str);
}
````

markaswift, 2016-10-10:
Is this fixed? I also have a W5100 gateway that I'd like 2 connection to.

henrikekblad, 2016-11-24:
Closing this until it's better supported by nativity by the Ethernet library.

closed 2016-11-24 by henrikekblad

NiklasOlofsson, 2017-04-05:
I also has the need for at least 2 connections. No news here?

fallberg, 2017-04-09:
It is possible to support infinite connections to a gw using node-red and socat. I use it with great success with serial gateway connected to a rPi running node-red and socat pipes which enable me to connect four different controllers to it simultaneously though tcp. No need to muck about in the embedded code. Not even ethernet is needed to the GW in this case.

NiklasOlofsson, 2017-04-09:
hehe, say what? ;)
Socat? Never used nodered but I know what it is. I think.

soif, 2017-04-10:
BTW I ended up using a NodeMCU as ~~Ethernet~~ Wifi Gateway. It support multiple connection, and is really cheap

fallberg, 2017-04-10:
@NiklasOlofsson 
https://www.mysensors.org/controller/nodered
https://forum.mysensors.org/topic/4599/how-to-add-a-serial-device-to-use-in-node-red