mysensors/MySensors #403: Force (manual) presentation is required!

maxtox, 2016-02-21:
Hello,

usualy a presentation made _only in the setup_, BUT sometimes f.e:

1. if GW crashes or gets an update with deleting DB of values/childid's
2. Controler crashes or gets an update with deleting DB of values/childid's

so you** can't** continue with setup in your controller, without restarting ALL of you (maybe hidden) nodes!

you need to force the presentation to get all available  sensors/nodes (at least which are not asleep).

Can you please include this future to be able to force presentation? like...

**I_REQUEST_PRESENTATION**

Thank you in Advance!

fallberg, 2016-02-21:
This is already implemented: https://github.com/mysensors/Arduino/blob/development/libraries/MySensors/core/MySensorCore.cpp#L288
Controller should just have to broadcast a I_PRESENTATION message and get presentations back.

closed 2016-02-21 by fallberg

maxtox, 2016-02-21:
Hello fallberg,

thank you!

i am in contact (supporting) the DEV of ioBroker, who is writing the new Controller for mysensors, which should implement "all" 1.6 features... dou you have a bit of informations about

I_HEARTBEAT, 
	I_PRESENTATION, 
	I_DISCOVER, 
	I_DISCOVER_RESPONSE, 
	I_HEARTBEAT_RESPONSE,
	I_LOCKED                //!< Node is locked (reason in string-payload)

Thank you in Advance!

fallberg, 2016-02-21:
@henrikekblad do we have doxygen in place for these? Or any other form of presentation? I have only covered security parts myself.