mysensors/MySensors #404: Esp8266Gateway wdt with latest IDE

biohazardxxx, 2016-02-23:
HI,

with latest ESP8266 2.1.0-rc2 and Arduino IDE 1.6.7 I can't get the Esp8266Gateway  sketch to run:

> ESP8266 MySensors Gateway
> Connecting to MyWiFi
> ......Connected!
> IP: 192.168.0.193
> 
>  ets Jan  8 2013,rst cause:4, boot mode:(3,7)
> 
> wdt reset
> load 0x4010f000, len 1264, room 16 
> tail 0
> chksum 0x42
> csum 0x42
> ~ld<LF>

I'm using ESP12 that runs fine all kind of other sketches.
Adding some serial debug lines tells that the reset happens at line:
  `gw.begin(incomingMessage, 0, true, 0);`
Best regards
-Alex

henrikekblad, 2016-02-28:
Can you reproduce this when using the latest in development branch?

biohazardxxx, 2016-02-29:
The dev branch looks quite different. Can't find the same example there. Only:
https://github.com/mysensors/Arduino/blob/development/libraries/MySensors/examples/GatewayESP8266/GatewayESP8266.ino
Is empty in Setup() and loop().

Using example from master gives error during compile:
Esp8266Gateway.ino:71:28: fatal error: MySigningNone.h: No such file or directory

henrikekblad, 2016-02-29:
That is the correct gateway sketch. 

You'll have to replace the whole MySensors library to be able to compile.

biohazardxxx, 2016-02-29:
ok got it compiled and uploaded. No resets so far. Thanks!

closed 2016-03-02 by henrikekblad