mysensors/MySensors #405: Ethernet W5100 Gateway closing TCP connection (2us- NOT 2ms) after each sending of data

maxtox, 2016-02-25:
checked yesterday and noticed, that if i use JUST STANDARD "TCP - W5100" Sketch, it happens that ...

(1) Connection is closing just (2 MICROseconds) after Arduino sends something.

(2) just after that  (2 MICROseconds) it starting to establish a new connection over the next TCP port, which is allways successful.

(3) After that it sends Data out and WITHOUT waiting of ACK or something else just closing again and 
starting by (1)

Checked everything with WIRESHARK with Windows and Linux (Raspi)!

WHEN I USE STANDARD ETHERNET sketch, from the Standard Arduino Library Example:

https://github.com/arduino/Arduino/tree/master/libraries/Ethernet

**everything is just fine! Means The connection ist establishing and staying open**

mp84, 2016-02-28:
Same here, while trying to connect to ioBroker.

henrikekblad, 2016-02-28:
Which version of MySensors are you running?

maxtox, 2016-02-28:
I am running 1.6

henrikekblad, 2016-02-28:
I don't get it. You're running the gateway in "server" mode right? That is, controller connects to the gateways ip.

And then you say that the gateway closes connection every time gateway is about to send something to controller?

How is the connection re-established? Does the controller re-connect?

When sending it basically does this line:
https://github.com/mysensors/Arduino/blob/development/libraries/MySensors/core/MyGatewayTransportEthernet.cpp#L183
That line shouldn't interrupt any connection unless it has been closed by controller and gateway notices this when trying to send.

Are you using dhcp? If yes, the lease could have expired and Ethernet.maintain() could be called. Unsure if that could close connection.
https://github.com/mysensors/Arduino/blob/development/libraries/MySensors/core/MyGatewayTransportEthernet.cpp#L373

maxtox, 2016-02-28:
Thank you for your Time...

i really appreceate that!!!!

UDP is Runnig JUST FINE!

You're running the gateway in "server" mode right? NO **GW running in CLIENT Mode.**.. so it connects to the Controller!

How is the connection re-established? Does the controller re-connect? YES but only if it GW closing the connection.

Are you using dhcp? NO DHCP

henrikekblad, 2016-02-28:
Ok, that make better sense. Client mode is pretty untested. Had a look at the code and it actually re-connects every send.

Could you try replacing he following block:
https://github.com/mysensors/Arduino/blob/development/libraries/MySensors/core/MyGatewayTransportEthernet.cpp#L158-L170

with
```
			#if defined(MY_CONTROLLER_URL_ADDRESS)
				if (client.connected() || client.connect(MY_CONTROLLER_URL_ADDRESS, MY_PORT)) {
			#else
				if (client.connected() || client.connect(_ethernetControllerIP, MY_PORT)) {
			#endif
				client.write(_ethernetMsg, strlen(_ethernetMsg));
			}
			else {
				// connecting to the server failed!
				ret = false;
			}
```

maxtox, 2016-02-28:
doing...

maxtox, 2016-02-28:
soooo hier the Log with the **NEW** .CPP

```
mysensors.0	2016-02-28 18:51:59	debug	mysensors.0 Send raw data: 0;2;0;0;1;
mysensors.0	2016-02-28 18:51:59	info	mysensors.0 Connected 192.168.111.99:49153
mysensors.0	2016-02-28 18:51:59	debug	inMem message mysensors.0.* mysensors.0.info.connection
mysensors.0	2016-02-28 18:51:59	error	mysensors.0 Cannot send to 192.168.111.99 because not connected
mysensors.0	2016-02-28 18:51:59	debug	mysensors.0 Send raw data: 0;0;3;0;19;force presentation
mysensors.0	2016-02-28 18:51:59	error	mysensors.0 Cannot send to 192.168.111.99 because not connected
mysensors.0	2016-02-28 18:51:59	debug	mysensors.0 Send raw data: 0;2;0;0;1;
mysensors.0	2016-02-28 18:51:59	error	mysensors.0 Error for "192.168.111.99": Error: read ECONNRESET
mysensors.0	2016-02-28 18:51:57	debug	mysensors.0 Sent to 192.168.111.99 0;0;3;0;13;force restart
mysensors.0	2016-02-28 18:51:57	debug	mysensors.0 Send raw data: 0;0;3;0;13;force restart
mysensors.0	2016-02-28 18:51:57	debug	mysensors.0 Sent to 192.168.111.99 0;0;3;0;13;force restart
mysensors.0	2016-02-28 18:51:57	debug	mysensors.0 Send raw data: 0;0;3;0;13;force restart
mysensors.0	2016-02-28 18:51:54	debug	inMem message mysensors.0.* mysensors.0.info.connection
mysensors.0	2016-02-28 18:51:54	debug	mysensors.0 Sent to 192.168.111.99 0;0;3;0;19;force presentation
mysensors.0	2016-02-28 18:51:54	debug	mysensors.0 Sent to 192.168.111.99 0;2;0;0;1;
mysensors.0	2016-02-28 18:51:54	debug	mysensors.0 Send raw data: 0;0;3;0;19;force presentation
mysensors.0	2016-02-28 18:51:54	debug	mysensors.0 Send raw data: 0;2;0;0;1;
mysensors.0	2016-02-28 18:51:54	info	mysensors.0 Connected 192.168.111.99:49153
mysensors.0	2016-02-28 18:51:54	debug	inMem message mysensors.0.* mysensors.0.info.connection
mysensors.0	2016-02-28 18:51:54	error	mysensors.0 Cannot send to 192.168.111.99 because not connected
mysensors.0	2016-02-28 18:51:54	debug	mysensors.0 Send raw data: 0;0;3;0;19;force presentation
mysensors.0	2016-02-28 18:51:54	error	mysensors.0 Cannot send to 192.168.111.99 because not connected
mysensors.0	2016-02-28 18:51:54	debug	mysensors.0 Send raw data: 0;2;0;0;1;
mysensors.0	2016-02-28 18:51:54	error	mysensors.0 Error for "192.168.111.99": Error: read ECONNRESET
mysensors.0	2016-02-28 18:51:52	debug	mysensors.0 Sent to 192.168.111.99 0;0;3;0;13;force restart
mysensors.0	2016-02-28 18:51:52	debug	mysensors.0 Send raw data: 0;0;3;0;13;force restart
mysensors.0	2016-02-28 18:51:52	debug	mysensors.0 Sent to 192.168.111.99 0;0;3;0;13;force restart
mysensors.0	2016-02-28 18:51:52	debug	mysensors.0 Send raw data: 0;0;3;0;13;force restart
mysensors.0	2016-02-28 18:51:49	debug	inMem message mysensors.0.* mysensors.0.info.connection
mysensors.0	2016-02-28 18:51:49	debug	mysensors.0 Sent to 192.168.111.99 0;0;3;0;19;force presentation
mysensors.0	2016-02-28 18:51:49	debug	mysensors.0 Sent to 192.168.111.99 0;2;0;0;1;
mysensors.0	2016-02-28 18:51:49	debug	mysensors.0 Send raw data: 0;0;3;0;19;force presentation
mysensors.0	2016-02-28 18:51:49	debug	mysensors.0 Send raw data: 0;2;0;0;1;
mysensors.0	2016-02-28 18:51:49	info	mysensors.0 Connected 192.168.111.99:49153
mysensors.0	2016-02-28 18:51:49	debug	inMem message mysensors.0.* mysensors.0.info.connection
mysensors.0	2016-02-28 18:51:49	error	mysensors.0 Cannot send to 192.168.111.99 because not connected
mysensors.0	2016-02-28 18:51:49	debug	mysensors.0 Send raw data: 0;0;3;0;19;force presentation
mysensors.0	2016-02-28 18:51:49	error	mysensors.0 Cannot send to 192.168.111.99 because not connected
mysensors.0	2016-02-28 18:51:49	debug	mysensors.0 Send raw data: 0;2;0;0;1;
mysensors.0	2016-02-28 18:51:49	error	mysensors.0 Error for "192.168.111.99": Error: read ECONNRESET
```

maxtox, 2016-02-28:
soooo here the Log with the **OLD** .CPP

192.168.111.99 = Arduino GW

every 5 sec it loosing Connection completely 

PORT of connection is changing!!!

```
mysensors.0	2016-02-28 18:55:26	warn	mysensors.0 Connection "192.168.111.99" closed unexpectedly
mysensors.0	2016-02-28 18:55:26	debug	inMem message mysensors.0.* mysensors.0.info.connection
mysensors.0	2016-02-28 18:55:26	debug	mysensors.0 TCP data received from 192.168.111.99:49233: 0;42;1;0;0;-127.00
mysensors.0	2016-02-28 18:55:25	info	mysensors.0 Connected 192.168.111.99:49233
mysensors.0	2016-02-28 18:55:25	debug	inMem message mysensors.0.* mysensors.0.192_168_111_99.0.58_TEMP.S_TEMP.V_TEMP
mysensors.0	2016-02-28 18:55:25	debug	inMem message mysensors.0.* mysensors.0.info.connection
mysensors.0	2016-02-28 18:55:25	debug	inMem message mysensors.0.* mysensors.0.info.connection
mysensors.0	2016-02-28 18:55:25	debug	mysensors.0 Set value Test NTC.V_TEMP 58: 38.5 number
mysensors.0	2016-02-28 18:55:25	debug	mysensors.0 Got: {"id":"0","childId":"58","type":"set","ack":false,"payload":"38.50","subType":"V_TEMP"}
mysensors.0	2016-02-28 18:55:25	warn	mysensors.0 Connection "192.168.111.99" closed unexpectedly
mysensors.0	2016-02-28 18:55:25	debug	mysensors.0 TCP data received from 192.168.111.99:49232: 0;58;1;0;0;38.50
```

maxtox, 2016-02-28:
WAIT!!!

i thing the new CPP is working...

Will tell you tomorrow if all ok!!!

maxtox, 2016-02-28:
after testing of the new .CPP

i noticed, that GW doesn't send anything! I mean NOTHING is comming to the Controller!

Serial Debug showing that it tries to send, but nothing is comming out of the Arduino.

When i switch to UDP

everything is just fine!

Look here...Made Video for you!

[]((https://youtu.be/ML7DtdM7tq0))

mp84, 2016-02-29:
Hmm, just testet this fix an seems to work for me!

maxtox, 2016-02-29:
will test in a minute...

maxtox, 2016-02-29:
@mp84 : please post the sketch here!

mp84, 2016-02-29:
It's pretty much the original GatewayW5100.ino sketch from the dev-branch with modified IP-Addresses and the patch by HEK in MyGatewayTransportEthernet.cpp

```
/**
 * The MySensors Arduino library handles the wireless radio link and protocol
 * between your home built sensors/actuators and HA controller of choice.
 * The sensors forms a self healing radio network with optional repeaters. Each
 * repeater and gateway builds a routing tables in EEPROM which keeps track of the
 * network topology allowing messages to be routed to nodes.
 *
 * Created by Henrik Ekblad <henrik.ekblad@mysensors.org>
 * Copyright (C) 2013-2015 Sensnology AB
 * Full contributor list: https://github.com/mysensors/Arduino/graphs/contributors
 *
 * Documentation: http://www.mysensors.org
 * Support Forum: http://forum.mysensors.org
 *
 * This program is free software; you can redistribute it and/or
 * modify it under the terms of the GNU General Public License
 * version 2 as published by the Free Software Foundation.
 *
 *******************************
 *
 * REVISION HISTORY
 * Version 1.0 - Henrik EKblad
 * Contribution by a-lurker and Anticimex,
 * Contribution by Norbert Truchsess <norbert.truchsess@t-online.de>
 * Contribution by Tomas Hozza <thozza@gmail.com>
 *
 *
 * DESCRIPTION
 * The EthernetGateway sends data received from sensors to the ethernet link.
 * The gateway also accepts input on ethernet interface, which is then sent out to the radio network.
 *
 * The GW code is designed for Arduino 328p / 16MHz.  ATmega168 does not have enough memory to run this program.
 *
 * LED purposes:
 * - To use the feature, uncomment WITH_LEDS_BLINKING in MyConfig.h
 * - RX (green) - blink fast on radio message recieved. In inclusion mode will blink fast only on presentation recieved
 * - TX (yellow) - blink fast on radio message transmitted. In inclusion mode will blink slowly
 * - ERR (red) - fast blink on error during transmission error or recieve crc error
 *
 * See http://www.mysensors.org/build/ethernet_gateway for wiring instructions.
 *
 */

// Enable debug prints to serial monitor
#define MY_DEBUG 

// Enable and select radio type attached
#define MY_RADIO_NRF24
//#define MY_RADIO_RFM69

// Enable gateway ethernet module type 
#define MY_GATEWAY_W5100

// W5100 Ethernet module SPI enable (optional if using a shield/module that manages SPI_EN signal)
//#define MY_W5100_SPI_EN 4  

// Enable Soft SPI for NRF radio (note different radio wiring is required)
// The W5100 ethernet module seems to have a hard time co-operate with 
// radio on the same spi bus.
#if !defined(MY_W5100_SPI_EN) && !defined(ARDUINO_ARCH_SAMD)
  #define MY_SOFTSPI
  #define MY_SOFT_SPI_SCK_PIN 14
  #define MY_SOFT_SPI_MISO_PIN 16
  #define MY_SOFT_SPI_MOSI_PIN 15
#endif  

// When W5100 is connected we have to move CE/CSN pins for NRF radio
#define MY_RF24_CE_PIN 5
#define MY_RF24_CS_PIN 6

// Enable to UDP
//#define MY_USE_UDP

#define MY_IP_ADDRESS 192,168,1,102   // If this is disabled, DHCP is used to retrieve address
// Renewal period if using DHCP
//#define MY_IP_RENEWAL_INTERVAL 60000
// The port to keep open on node server mode / or port to contact in client mode
#define MY_PORT 5003      

// Controller ip address. Enables client mode (default is "server" mode). 
// Also enable this if MY_USE_UDP is used and you want sensor data sent somewhere. 
#define MY_CONTROLLER_IP_ADDRESS 192, 168, 1, 11   
 
// The MAC address can be anything you want but should be unique on your network.
// Newer boards have a MAC address printed on the underside of the PCB, which you can (optionally) use.
// Note that most of the Ardunio examples use  "DEAD BEEF FEED" for the MAC address.
#define MY_MAC_ADDRESS 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED

// Flash leds on rx/tx/err
#define MY_LEDS_BLINKING_FEATURE
// Set blinking period
#define MY_DEFAULT_LED_BLINK_PERIOD 300

// Enable inclusion mode
#define MY_INCLUSION_MODE_FEATURE
// Enable Inclusion mode button on gateway
#define MY_INCLUSION_BUTTON_FEATURE
// Set inclusion mode duration (in seconds)
#define MY_INCLUSION_MODE_DURATION 60 
// Digital pin used for inclusion mode button
#define MY_INCLUSION_MODE_BUTTON_PIN  3 

// Uncomment to override default HW configurations
//#define MY_DEFAULT_ERR_LED_PIN 7  // Error led pin
//#define MY_DEFAULT_RX_LED_PIN  8  // Receive led pin
//#define MY_DEFAULT_TX_LED_PIN  9  // the PCB, on board LED

#include <SPI.h>

#if defined(MY_USE_UDP)
  #include <EthernetUdp.h>
#endif
#include <Ethernet.h>
#include <MySensor.h>

void setup()
{
}

void loop() {
}
```

BTW: Any chanse to contact you directly? Got some questions relating to the iobroker.mysensors adapter...

maxtox, 2016-02-29:
yes... contact me  in iobroker forum my nick is: maxtox

Hoe do you see that it is working? you do not send anything?!

mp84, 2016-02-29:
I'm monitoring the connection with TCPDump.
And i'm sending values trough ioBroker to my Dimmer-Node and the Light turns on and off... That's how i see it's working :)

maxtox, 2016-02-29:
IT IS NOT WORKING!

```
....

void setup()
{
}

void loop() {
  for (int i = 0; i < 20; i++){

    MyMessage msg(i, 3);
    Serial.println("i want to send");
    send(msg.setSensor(i).set(1, 2), false);

    delay (1000);
  }}

```
even this "stupid" sketch does not send anything :-(

mp84, 2016-02-29:
```
mysensors.0	2016-02-29 21:01:26	debug	inMem message mysensors.0.* mysensors.0.172_16_1_102.21.0_DIMMER.S_DIMMER.V_PERCENTAGE
mysensors.0	2016-02-29 21:01:26	debug	mysensors.0 Set value Bürolicht 0: 25 string
mysensors.0	2016-02-29 21:01:26	debug	mysensors.0 Got: {"id":"21","childId":"0","type":"set","ack":false,"payload":"25","subType":"V_DIMMER"}
mysensors.0	2016-02-29 21:01:26	debug	mysensors.0 TCP data received from 172.16.1.102:49153: 21;0;1;0;3;25
mysensors.0	2016-02-29 21:01:26	debug	inMem message mysensors.0.* mysensors.0.172_16_1_102.21.0_DIMMER.S_DIMMER.V_STATUS
mysensors.0	2016-02-29 21:01:26	debug	mysensors.0 Set value 172_16_1_102.21.0_DIMMER.S_DIMMER.V_STATUS 0: true boolean
mysensors.0	2016-02-29 21:01:26	debug	mysensors.0 Got: {"id":"21","childId":"0","type":"set","ack":false,"payload":"1","subType":"V_LIGHT"}
mysensors.0	2016-02-29 21:01:26	debug	mysensors.0 TCP data received from 172.16.1.102:49153: 21;0;1;0;2;1
mysensors.0	2016-02-29 21:01:25	debug	mysensors.0 Sent to 172.16.1.102 21;0;1;0;3;25
mysensors.0	2016-02-29 21:01:25	debug	mysensors.0 Send raw data: 21;0;1;0;3;25
mysensors.0	2016-02-29 21:01:25	debug	mysensors.0 stateChange mysensors.0.172_16_1_102.21.0_DIMMER.S_DIMMER.V_PERCENTAGE {"val":25,"ack":false,"ts":1456776086,"q":0,"from":"system.adapter.web.0","lc":1456776086}
mysensors.0	2016-02-29 21:01:25	debug	inMem message mysensors.0.* mysensors.0.172_16_1_102.21.0_DIMMER.S_DIMMER.V_PERCENTAGE
mysensors.0	2016-02-29 21:01:22	info	mysensors.0 Received INTERNAL message: I_SKETCH_VERSION: 1.1
mysensors.0	2016-02-29 21:01:22	debug	mysensors.0 Got: {"id":"21","childId":"255","type":"internal","ack":false,"payload":"1.1","subType":"I_SKETCH_VERSION"}
mysensors.0	2016-02-29 21:01:22	debug	mysensors.0 TCP data received from 172.16.1.102:49153: 21;255;3;0;12;1.1
mysensors.0	2016-02-29 21:01:22	info	mysensors.0 Name from 172.16.1.102 :DimmableLED
mysensors.0	2016-02-29 21:01:22	debug	mysensors.0 Got: {"id":"21","childId":"255","type":"internal","ack":false,"payload":"DimmableLED","subType":"I_SKETCH_NAME"}
mysensors.0	2016-02-29 21:01:22	debug	mysensors.0 TCP data received from 172.16.1.102:49153: 21;255;3;0;11;DimmableLED
mysensors.0	2016-02-29 21:01:22	debug	mysensors.0 Got: {"id":"21","childId":"0","type":"presentation","ack":false,"payload":"","subType":"S_DIMMER"}
mysensors.0	2016-02-29 21:01:22	debug	mysensors.0 TCP data received from 172.16.1.102:49153: 21;0;0;0;4;
mysensors.0	2016-02-29 21:01:20	info	mysensors.0 Config from 172.16.1.102 :Imperial
mysensors.0	2016-02-29 21:01:20	debug	mysensors.0 Got: {"id":"21","childId":"255","type":"internal","ack":false,"payload":"0","subType":"I_CONFIG"}
mysensors.0	2016-02-29 21:01:20	debug	mysensors.0 TCP data received from 172.16.1.102:49153: 21;255;3;0;6;0
mysensors.0	2016-02-29 21:01:20	debug	mysensors.0 Got: {"id":"21","childId":"255","type":"presentation","ack":false,"payload":"2.0.0-beta","subType":"S_ARDUINO_NODE"}
mysensors.0	2016-02-29 21:01:20	debug	mysensors.0 TCP data received from 172.16.1.102:49153: 21;255;0;0;17;2.0.0-beta
mysensors.0	2016-02-29 21:01:20	debug	mysensors.0 Got: {"id":"21","childId":"0","type":"req","ack":false,"payload":"","subType":"V_DIMMER"}
mysensors.0	2016-02-29 21:01:20	debug	mysensors.0 TCP data received from 172.16.1.102:49153: 21;0;2;0;3;
```

Try resetting any of your nodes, the gateway should then open a connection to the broker and transmit data. Just testet an realised that the boker cannot establish connection to the gateway. Connection has to be initiated by the gateway...

maxtox, 2016-02-29:
NOW it is WORKING!!!!

ONLY WITH THE NEW CPP

@henrikekblad could you please make a pullrequest?! 

Thank you!!!!

henrikekblad, 2016-03-01:
Ok good.

mp84, 2016-03-01:
@maxtox did you change something or just reset a node?

maxtox, 2016-05-05:
Changes are still not in this branch.
Here is PR https://github.com/mysensors/Arduino/pull/430

closed 2016-05-08 by henrikekblad

JohanElmis, 2016-08-05:
Hey, 

Have you  read the latest article written by me? Just take  a look, it's very interesting. Please read  here <http://mark.4bikerzonly.com/e4spxszh>

Warmest regards, johan