mysensors/MySensors #408: Switch radio SCN default pin for ESP8266

henrikekblad, 2016-03-02:
http://forum.mysensors.org/topic/3285/solved-esp8266-fails-to-boot-with-nrf24l01-connected/5

From @Daanoz debugging:
------------------------------------------
Ok, after debugging for a while: the problem is with the CSN pin on GPIO pin 15. Pin 15 is also used in the ESP8266 to define boot from SD card. Somehow, at least in my case, after a reset, the NRF chip sets the CSN pin to high (measured it with a multimeter), causing the ESP8266 to start the SD-card procedure, which fails of course. The solution for me is swapping the pins for inclusion with the CSN pin:

So:
CSN -> GPIO 5 -> D1
Inclusion mode -> GPIO 15 -> D7

Yveaux, 2016-03-02:
Which ESP board are we discussing here?
The NodeMcu board, as described in the official ESP build description on MySensors.org uses D8 for CSN.

daanoz, 2016-03-02:
Sorry, I made a typo... I had the problem with the D8 pin... (GPIO 15)

emc2cube, 2016-03-02:
If the default wiring for the ESP8266 gateway end up being modified for 2.0, please consider moving CE (D2, GPIO4) away to D4, D9 or D10 to leave GPIO 4&5 free for I2C devices.
With the ability to add sensors directly on the gateway this would be a game changer. (Else we have to declare the I2C bus on the sketch, which can be done but is not as user friendly and may interfere with debug via serial monitoring https://github.com/emc2cube/MyWeatherGatewayESP8266/blob/master/MyWeatherGatewayESP8266.ino )

closed 2016-11-24 by henrikekblad

877dev, 2023-04-29:
Thanks guys for this issue, I have managed to get my Wemos D1 mini ESP8266 working with an SPI ethernet shield (W5500) after the same issues. The solution was using GPIO5 instead of GPIO15 as stated.

Wiring:
    GPIO    ESP8266  Name     ===================================
     12           D6      MISO 
     13           D7      MOSI
     5             D1       SS 
     14           D5      SCK

Here's the full test sketch defining the SS pin as GPIO5/D1 if anyone searches for this:
```
#include <SPI.h>
#include <Ethernet.h>

byte mac[] = {  0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };
IPAddress ip(192,168,2,2);

void setup() {
  Serial.begin(74880);

  Ethernet.init(5);  // ESP8266 set SS pin

  // disable SD card if one in the slot
  pinMode(4,OUTPUT);
  digitalWrite(4,HIGH);

  Serial.println("Starting w5100");
  Ethernet.begin(mac,ip);

  Serial.println(Ethernet.localIP());
}

void loop() {
}
```