mysensors/MySensors #410: st=fail

trlafleur, 2016-03-04:
From time to time, I get st=fail, if I put a wait(200) after each send() they are gone...
using 2.0.0 beta

There must be some timing issue in the stack, one should NOT need to add the wait() to the code to fix something like this...

I always get an st=fail with  on statup, might be related:

send: 6-6-0-0 s=255,c=3,t=15,pt=0,l=2,sg=0,st=ok:
send: 6-6-0-0 s=255,c=0,t=18,pt=0,l=10,sg=0,st=fail:2.0.0-beta  <-------------
send: 6-6-0-0 s=255,c=3,t=6,pt=1,l=1,sg=0,st=ok:0

markaswift, 2016-04-19:
I believe I'm seeing the same thing...

henrikekblad, 2016-04-19:
I usually only see these type of ack-fail when under-powering the receiving sides radio.
 
Adding a capacitor usually helps. But I guess you've already tried this and feeding radio from something more powerful than the 3v3 rail of the Arduino right?

trlafleur, 2016-04-23:
My setup is multiple Moteino each with 100uf/10v 0805 chip capacitor soldered direct to the radio power pin to gnd. Power to radio is via and external 3.3v power source with lots of current available.  

Without adding wait(100) prior each send, I get this often.... So it looks to me to be a stack timing related issue with these radios.

henrikekblad, 2016-04-24:
http://forum.mysensors.org/topic/3663/testing-development-branch-with-rf69hw-is-not-working-as-it-should

PR #437 (2016-05-22): update RFM69 driver
- fixes #436
- fixes #410
- fixes #407
closed 2016-05-22 by tekka007