mysensors/MySensors #421: Radio Packet Size

trlafleur, 2016-04-10:
It appears that the max packet size is limited by the radio type. In a given system, one uses only one radio type, so:

1) system packed size should be set by radio type or config file option, allowing user of RFM69 radios to use a larger payload.
2) Future support for multi packed delivery of payload. This will alow any radio type to send longer payload message

fallberg, 2016-04-10:
I recommend reject. The signing infrastructure will not support this and it will make radio agnostic design very hard.

closed 2016-04-19 by henrikekblad

trlafleur, 2016-04-23:
As the evolution of this project  move forward we will see new radios and techniques, so maybe it's time to be able to can figure size of the radio message based on the radio use and or allow for multiple packets to be set up .....we need to look forward to the future

fallberg, 2016-04-23:
You already have this support. For rf69 max size is 64, for RF24 it is 32.

fallberg, 2016-04-23:
Sorry, my bad, it is fixed to 32 bytes.
But you have to remember, goal is to keep codebase reasonably complex as it needs to fit resource limited devices. It would kill the project if we got to the point where existing hw deployed in various homes would not be able to contain the library due to size restrictions. 
Of course, the features you seek could be controlled by a flag, but all nodes in the chain would need to support the same features (gateways and repeaters).
But it is an open source project. Everybody are welcome to contribute so if you have an idea on how to do this without adding considerable size to the codebase, you are welcome to file a pull request.