mysensors/MySensors #436: Using Current RFM69 driver

trlafleur, 2016-05-07:
I have started using the newer RFM69 driver from LowPowerLab, this fix my occasional ST-fail issue with this radio and allows me to remove all of my wait(200) after each send. I'm also informed, that it may fixes the problem with the Ethernet gateway sharing of the hardware SPI, but I have not tested yet...

These are the ONLY changes that I made to there files:
In RFM69.cpp

from:
#include <RFM69.h>
#include <RFM69registers.h>
#include <SPI.h>

To:
#include "RFM69.h"
#include "RFM69registers.h"
#include <SPI.h>

https://github.com/LowPowerLab/RFM69

henrikekblad, 2016-05-08:
The comment looks a bit weird (#include <nothing>). 

If you've identified any problems, could you prepare a pull request with the changes?

trlafleur, 2016-05-08:
I fixed the includes... (did not have the ~~~)
I just replaced the three files in RFM69 folder with the new files from LowPowerLabs, made the two changes, done...

I'm not set up to do PULL request...

trlafleur, 2016-05-10:
I have all this working now and have 7 devices on it to a serial gateway...
Using new (current) RFM69 driver and RFM69_ATC...
Its interesting to see the power levels change as packets flow...

If I can figure out how to do a PULL request, I will make it happen to 2.0b development branch...

tom

PR #437 (2016-05-22): update RFM69 driver
- fixes #436
- fixes #410
- fixes #407
closed 2016-05-22 by tekka007