mysensors/MySensors #466: Sensebender Micro, (Battery) Voltage sending very wrong...?

gizmocuz, 2016-06-11:
Hi (Thomas),

The value being send seems not correct, and needs to be divided by 1000.0

This can also be seen in the sketch code, where it subtract 1900 (1 9 0 0 ) from it (1.9 volt)

I think the sending part should become:

```
#ifdef BATT_SENSOR
    float send_voltage = float(vcc)/1000.0f;
    send(msgBatt.set(send_voltage,3));
#endif
```

Before the Change: (current implementation):
[image]

After the Change:
[image]

Am i correct? Shall i make a pull request ?

tbowmo, 2016-06-12:
Yes, you are right.. Can't remember why I didn't make it a floating value back then? Perhaps it was just for quick tests..

Please create a PR for it.

gizmocuz, 2016-06-12:
I made a pull request #469 
But somehow it got errors, dont know why... looking all good ?

fallberg, 2016-06-12:
Your commit message headline is too long. Amend it and shorten it to < 50 chars. Check the link posted by Jenkins/Gitler. You should also have received a mail.

gizmocuz, 2016-06-12:
I give up !, Made a new Pull request, with a shorter name #471 
Dont know what is wrong with it, and it makes me VERY irritating !... If this continues, i wont submit any patch/addon anymore!
What is possible wrong with this ? Shortend the message....

gizmocuz, 2016-06-12:
Maybe we can submit a pull request that gets rid of this check ?

fallberg, 2016-06-12:
Like I said, details are mailed to the author. If you can't read the mail registered to your github account, read this: http://chris.beams.io/posts/git-commit
Follow that and your commits will be accepted.
At least we don't rant on peoples forum signatures...

fallberg, 2016-06-12:
I also think in the most recent case, that the git commit complaints are stemming from the parent commit of your pr. And that parent is not the most recent or reasonably recent because it is not possible to merge the commit on the Web, which indicates it needs to be rebased. And that will probably also cause the git commit complaint to dissappear. So just pull the development branch to your fork, rebase your topic branch and force push it to github and it should pass.

gizmocuz, 2016-06-13:
New try... hope this one works now...

closed 2016-06-13 by tekka007

JohanElmis, 2016-08-05:
Hello, 

I wanted to  share  with you my impressions about a book I've  just finished reading,  it's awesome, you can find my  review here <http://super.chaiparty.com/e4vdzgw>

Good wishes, johan

JohanElmis, 2016-09-01:
Yo! 

I wanted to ask you whether you have  any previous experience of this type  of work? Here are  some questions for you <http://chiltunemy.eduardopolanco.com/e4unq>

Speak to you later, johan

JohanElmis, 2016-09-01:
Hello, 

During my last journey I've visited a wonderful place, you'd  love it! Please take a look here <http://fivytuki.eds-homes.com/e4apyces>

Good wishes, johan