mysensors/MySensors #468: Improve smartSleep: add a "no more commands" message to the gateway

abmantis, 2016-06-11:
Currently, smartSleep() simply waits a specified amount of time for any command from the gateway.
Instead, after sending the heartbeat, the node could wait (with a timeout value) for a "no more commands" message from the gateway and sleep after receiving that. This would reduce the waiting time before sleeping.
Another improvement would be having a "request commands" from the node instead of the heartbeat.

So, on the node:
- Send "request commands"
- Wait until "no more commands" is received
- Sleep

On the gateway:
- "request commands" is received from a node
- Send commands to node
- Send "no more commands" message.

MartinHjelmare, 2016-08-01:
This is good. I assume the gateway in the above description means the controller.

tekka007, 2017-03-29:
see #722, use dedicated messages

closed 2017-03-29 by tekka007