mysensors/MySensors #485: Issues if MySensor.h gets included in sub-library

DerTomm, 2016-06-24:
Hi MySensors Team,

thanks for this great library.
After some playing around I started to develop software for my sensor nodes with the MySensors lib. Today I realized a problem which occurs when I am including "MySensor.h" within a class in a private library in one of my projects (because there I process a given MyMessage object as input parameter of one method). So MySensors.h is used twice - once in the main sensor node cpp and once in this library class. The result is a compiler error

```
In file included from src\PiduinoCommons\SensorNode.h:12:0,
from src\PiduinoCommons\SensorNode.cpp:8:
D:\Development\Arduino\PiduinoNetwork\libs\MySensors_Development\MySensors/MySensor.h:287:4: error: #error No forward link or gateway feature activated. This means nowhere to send messages! Pretty pointless.
#error No forward link or gateway feature activated. This means nowhere to send messages! Pretty pointless."
```
Obviously the second include gets processed first - before the "MY_"-defines of the main class are parsed.
Is there any chance to include MySensor.h in other source classes without this behaviour?

wmarkow, 2016-07-13:
Hello @DerTomm. It is probably because of "scope of visibility" (called [translation unit\http://stackoverflow.com/questions/1106149/what-is-a-translation-unit-in-c) your defines.  If I've understood correctly, in order to provide your own configurations of MySensor library, you need to put some defines in your local *.cpp file and then include <MySensors.h>, like this for example:
```
#define MY_RADIO_NRF24
#define MY_RF24_PA_LEVEL RF24_PA_LOW
#define MY_GATEWAY_SERIAL

#include <MySensors.h> 
```
Your definitions will be visible only when compiling the current cpp file. If you include <MySensors.h> in other cpp file as well, then the MySensors.h will complain that some defines are unknown - as there is a different  translation unit. To overcome this you need to put the same defines again in the second file; or even better you can create your own configuration MyOwnConfig.h, put the defines there and include it always like this:
```
#include <MyOwnConfig.h> 
#include <MySensors.h> 
```

DerTomm, 2016-07-13:
Thanks for these information. I finally managed to solve the issue with a forward declaration. Now the header files are used in the correct order so that MySensor.h sees its own definements.

closed 2016-07-13 by DerTomm

JohanElmis, 2016-09-01:
Hello! 

Just wanted to  say hello to you and let you know that I found something interesting for you, look  at it here <http://spuphultunge.mblvid.com/e4bxs>

Rushing, johan