mysensors/MySensors #491: Error compiling GatewayESP8266MQTTClient

supoeric, 2016-06-29:
when compile GatewayESP8266MQTTClient, I got error.
#include <core_version.h>:No such file or directory
#include "Schedule.h":No such file or directory

CynthiaGreen, 2016-06-29:
The latest development commit requires the ESP8266 2.3.0 Framework. 

Even then the sketch compiles but doesn't seem to actually connect to MQTT or run the setup/loop code. 

The output from the gateway is just a bunch of dots:

sta config unchangedf r-80, scandone
...............pm open,type:2 0
....................................................................................................................................................................................................

supoeric, 2016-07-01:
after update the libraries, I got 
"dhcp client start...
0;255;3;9;TSM:OK
f r0, scandone
................pm open, type:2 0
........................................................"

tekka007, 2016-07-01:
@supoeric the PR is not merged yet

closed 2016-07-02 by tekka007

Iodine-, 2016-07-19:
Has this been resolved?  I have 2.0 and still get

`connected with xxxxxx, channel 11
dhcp client start...
0;255;3;0;9;TSM:READY
f r0, scandone
...................pm open,type:2 0
.....................................`

and just infinite dots.

JohanElmis, 2016-08-05:
Hello friend, 

I've been looking for some stuff when by chance I found  that! It's so  sad,  please read ant tell me what you think  <http://shop.954homeloans.com/e4ixwt>

All the best, johan

JohanElmis, 2016-09-01:
Dear, 

I saw  you last achievements,  and I  think you need to keep up the good work! Here  is  some helpful  information  for you  <http://slopejova.starkmedical.net/e4tab>

johan

PR #497 (2016-07-02): Cleanup MyGatewayXXX and update PubSubClient
- Refactor names
- Add DHCP failure message
- Update PubSubClient to 2.4
- Fixes #491