mysensors/MySensors #509: 2.0.0 Stable Node tries to find parent although MY_PARENT_NODE_ID is given.

rollercontainer, 2016-07-11:
I've got 2.0.0 stable MqttClientGateway and a simple 2.0.0 stable node with MY_PARENT_NODE_ID set:
```
#define MY_DEBUG
#define MY_RADIO_NRF24
#define MY_NODE_ID 10
#define MY_PARENT_NODE_ID 0
#define MY_PARENT_NODE_IS_STATIC
```

It seems to ignore that and tries to find a parent:

```
Starting sensor (RNNNA-, 2.0.0)
TSM:INIT
TSM:RADIO:OK
TSP:ASSIGNID:OK (ID=10)
TSM:FPAR
TSP:MSG:SEND 10-10-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
TSM:FPAR
TSP:MSG:SEND 10-10-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
TSM:FPAR
TSP:MSG:SEND 10-10-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
TSM:FPAR
TSP:MSG:SEND 10-10-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
!TSM:FPAR:FAIL
!TSM:FAILURE
TSM:PDT
```

tekka007, 2016-07-11:
Thanks for reporting. 
Defining `MY_PARENT_NODE_ID` sets a preferred parent for routing/forwarding messages, if the parent is offline or not responding, the fallback mechanism kicks in and searches for a new (active) parent. 
Setting `MY_PARENT_NODE_IS_STATIC` should override this behavior and prevent parent node searchings and reassignments, this seems not to be the case and requires further investigations.

tekka007, 2016-07-12:
@rollercontainer Would be great if you could test this fix and report back: https://github.com/tekka007/MySensors/tree/StaticParents
Thanks!

rollercontainer, 2016-07-13:
Do I have to reflash the gateway too, or just the node?

tekka007, 2016-07-13:
Ideally, reflash both with the same code base.

rollercontainer, 2016-07-13:
Flashed both, gw and node with branch StaticParent.
With #define MY_PARENT_NODE_IS_STATIC node hangs in a loop of:
`
TSM:FPAR
TSM:FPAR:STATP
TSM:ID
TSM:CHKID:OK,ID=10
TSM:UPL
TSP:PING:SEND,DST=0
TSP:MSG:SEND 10-10-0-0 s=255,c=3,t=24,pt=1,l=1,sg=0,ft=0,st=ok:1
TSP:CHKUPL:FAIL
!TSM:UPL:FAIL
`
without it looks like this:
`
Starting sensor (RNNNA-, 2.0.0-beta)
TSM:INIT
TSM:RADIO:OK
TSP:ASSIGNID:OK,ID=10
TSM:FPAR
TSP:MSG:SEND 10-10-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
!TSM:FPAR:NOR
TSM:FPAR
TSP:MSG:SEND 10-10-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
!TSM:FPAR:NOR
TSM:FPAR
TSP:MSG:SEND 10-10-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
!TSM:FPAR:NOR
TSM:FPAR
TSP:MSG:SEND 10-10-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
!TSM:FPAR:FAIL
!TSM:FAILURE
TSM:PDT
`

tekka007, 2016-07-13:
Ok, the parent node (i.e. GW in your case) does not respond. 
Please add the full debug log of GW and the node and make sure, they are not too close to each other (possible radio interference)

rollercontainer, 2016-07-13:
Well, they are 2m away from eachother. Worked with 1.5 and 2.0.0beta...

GW says: 
0;255;3;0;9;Starting gateway (RNNGA-, 2.0.0-beta)
0;255;3;0;9;TSM:INIT
0;255;3;0;9;TSM:RADIO:OK
0;255;3;0;9;TSM:GW MODE
0;255;3;0;9;TSM:TSP READY
IP: 192.168.178.3
0;255;3;0;9;No registration required
0;255;3;0;9;Init complete, id=0, parent=0, distance=0, registration=1
IP: 192.168.178.3
0;255;3;0;9;Attempting MQTT connection...
0;255;3;0;9;MQTT connected
0;255;3;0;9;Sending message on topic: mysensors-out/0/1/1/0/0
0;255;3;0;9;Sending message on topic: mysensors-out/0/0/1/0/1
0;255;3;0;9;TSP:MSG:READ 10-10-255 s=255,c=3,t=7,pt=0,l=0,sg=0:
0;255;3;0;9;TSP:MSG:BC
0;255;3;0;9;TSP:MSG:FPAR REQ,ORI=10
0;255;3;0;9;TSP:CHKUPL:OK,FLDCTRL
0;255;3;0;9;TSP:MSG:GWL OK
0;255;3;0;9;!TSP:MSG:SEND 0-0-10-10 s=255,c=3,t=8,pt=1,l=1,sg=0,ft=0,st=fail:0
0;255;3;0;9;TSP:MSG:READ 10-10-255 s=255,c=3,t=7,pt=0,l=0,sg=0:
0;255;3;0;9;TSP:MSG:BC
0;255;3;0;9;TSP:MSG:FPAR REQ,ORI=10
0;255;3;0;9;TSP:CHKUPL:OK,FLDCTRL
0;255;3;0;9;TSP:MSG:GWL OK
0;255;3;0;9;!TSP:MSG:SEND 0-0-10-10 s=255,c=3,t=8,pt=1,l=1,sg=0,ft=0,st=fail:0
0;255;3;0;9;TSP:MSG:READ 10-10-255 s=255,c=3,t=7,pt=0,l=0,sg=0:
0;255;3;0;9;TSP:MSG:BC
0;255;3;0;9;TSP:MSG:FPAR REQ,ORI=10
0;255;3;0;9;TSP:CHKUPL:OK,FLDCTRL
0;255;3;0;9;TSP:MSG:GWL OK
0;255;3;0;9;!TSP:MSG:SEND 0-0-10-10 s=255,c=3,t=8,pt=1,l=1,sg=0,ft=0,st=fail:0
0;255;3;0;9;TSP:MSG:READ 10-10-255 s=255,c=3,t=7,pt=0,l=0,sg=0:
0;255;3;0;9;TSP:MSG:BC
0;255;3;0;9;TSP:MSG:FPAR REQ,ORI=10
0;255;3;0;9;TSP:CHKUPL:OK,FLDCTRL
0;255;3;0;9;TSP:MSG:GWL OK
0;255;3;0;9;!TSP:MSG:SEND 0-0-10-10 s=255,c=3,t=8,pt=1,l=1,sg=0,ft=0,st=fail:0

rollercontainer, 2016-07-13:
placed the node in the bedroom, approx. 6m distance

rollercontainer, 2016-07-13:
Its a node with FastLED lib, is that interferring the mysensor lib maybe? (Again: worked with beta)

tekka007, 2016-07-13:
Your node doesn't seem to receive messages sent from the GW:
`0;255;3;0;9;!TSP:MSG:SEND 0-0-10-10 s=255,c=3,t=8,pt=1,l=1,sg=0,ft=0,st=fail:0`
Maybe it is a power issue - 2.0.0 exchanges more messages during booting than 1.5 and earlier 2.0.0-beta versions. I suggest you try following: (1) replace/add a larger cap to the radio, (2) change node id, (3) upload simple sketch with dynamic ID. In all scenarios, use the latest 2.0.0 release for both GW and node and post the debug log here. Thanks for your help.

tekka007, 2016-07-15:
@rollercontainer Any updates on the issue?

rollercontainer, 2016-07-15:
Was too tired yesterday, will try it this afternoon.

rollercontainer, 2016-07-15:
Don't ask me, but now its working. I did nothing to it.

rollercontainer, 2016-07-15:
cap at radio was already added.

tekka007, 2016-07-15:
Sounds like radio interference or other HW-related issues

mahmoud-samy, 2016-07-27:
I have similar issue:

```
#define MY_DEBUG
#define MY_RADIO_NRF24
#define MY_RF24_PA_LEVEL  RF24_PA_LOW
```

> Starting sensor (RNNNA-, 2.0.1-beta)
> TSM:INIT
> TSM:RADIO:OK
> TSM:FPAR
> TSP:MSG:SEND 255-255-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
> TSM:FPAR
> TSP:MSG:SEND 255-255-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
> TSM:FPAR
> TSP:MSG:SEND 255-255-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
> TSM:FPAR
> TSP:MSG:SEND 255-255-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
> !TSM:FPAR:FAIL
> !TSM:FAILURE
> TSM:PDT
> TSM:INIT
> TSM:RADIO:OK
> TSM:FPAR
> TSP:MSG:SEND 255-255-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
> TSM:FPAR
> TSP:MSG:SEND 255-255-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
> TSM:FPAR
> TSP:MSG:SEND 255-255-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
> TSM:FPAR
> TSP:MSG:SEND 255-255-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
> !TSM:FPAR:FAIL
> !TSM:FAILURE
> TSM:PDT
> TSM:INIT
> TSM:RADIO:OK
> TSM:FPAR
> TSP:MSG:SEND 255-255-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
> TSM:FPAR
> TSP:MSG:SEND 255-255-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
> TSM:FPAR
> TSP:MSG:SEND 255-255-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
> TSM:FPAR
> TSP:MSG:SEND 255-255-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
> !TSM:FPAR:FAIL
> !TSM:FAILURE
> TSM:PDT
> TSM:INIT
> TSM:RADIO:OK
> TSM:FPAR
> TSP:MSG:SEND 255-255-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
> TSM:FPAR
> TSP:MSG:SEND 255-255-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
> TSM:FPAR
> TSP:MSG:SEND 255-255-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
> TSM:FPAR
> TSP:MSG:SEND 255-255-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
> !TSM:FPAR:FAIL
> !TSM:FAILURE
> TSM:PDT
>

mahmoud-samy, 2016-07-27:
After adding: 
```
#define MY_NODE_ID 10
#define MY_PARENT_NODE_ID  0
#define MY_PARENT_NODE_IS_STATIC
```
same issue

mahmoud-samy, 2016-07-27:
After switching to https://github.com/tekka007/MySensors/tree/StaticParents

> Starting sensor (RNNNA-, 2.0.0-beta)
> TSM:INIT
> TSM:INIT:TSP OK
> TSM:INIT:STATID,ID=10
> TSF:ASID:OK,ID=10
> TSM:FPAR
> TSM:FPAR:STATP
> TSM:ID
> TSM:ID:OK,ID=10
> TSM:UPL
> TSF:PING:SEND,TO=0
> !TSF:MSG:SEND,10-10-0-0,s=255,c=3,t=24,pt=1,l=1,sg=0,ft=0,st=NACK:1
> TSF:CHKUPL:FAIL
> !TSM:UPL:FAIL
> TSM:FPAR
> TSM:FPAR:STATP
> TSM:ID
> TSM:ID:OK,ID=10
> TSM:UPL
> TSF:PING:SEND,TO=0
> !TSF:MSG:SEND,10-10-0-0,s=255,c=3,t=24,pt=1,l=1,sg=0,ft=1,st=NACK:1
> TSF:CHKUPL:FAIL
> !TSM:UPL:FAIL
> TSM:FPAR
> TSM:FPAR:STATP
> TSM:ID
> TSM:ID:OK,ID=10
> TSM:UPL
> TSF:PING:SEND,TO=0
> !TSF:MSG:SEND,10-10-0-0,s=255,c=3,t=24,pt=1,l=1,sg=0,ft=2,st=NACK:1
> TSF:CHKUPL:FAIL
> !TSM:UPL:FAIL
> TSM:FPAR
> TSM:FPAR:STATP
> TSM:ID
> TSM:ID:OK,ID=10
> TSM:UPL
> TSF:PING:SEND,TO=0
> !TSF:MSG:SEND,10-10-0-0,s=255,c=3,t=24,pt=1,l=1,sg=0,ft=3,st=NACK:1
> TSF:CHKUPL:FAIL
> !TSM:UPL:FAIL
> TSM:FPAR
> TSM:FPAR:STATP
> TSM:ID
> TSM:ID:OK,ID=10
> TSM:UPL
> TSF:PING:SEND,TO=0
> !TSF:MSG:SEND,10-10-0-0,s=255,c=3,t=24,pt=1,l=1,sg=0,ft=4,st=NACK:1
> TSF:CHKUPL:FAIL
> !TSM:UPL:FAIL
> TSM:FPAR
> TSM:FPAR:STATP
> TSM:ID
> TSM:ID:OK,ID=10
> TSM:UPL
> TSF:PING:SEND,TO=0
> !TSF:MSG:SEND,10-10-0-0,s=255,c=3,t=24,pt=1,l=1,sg=0,ft=5,st=NACK:1
> TSF:CHKUPL:FAIL
> !TSM:UPL:FAIL
> TSM:FPAR
> TSM:FPAR:STATP
> TSM:ID
> TSM:ID:OK,ID=10
> TSM:UPL
> TSF:PING:SEND,TO=0
> !TSF:MSG:SEND,10-10-0-0,s=255,c=3,t=24,pt=1,l=1,sg=0,ft=6,st=NACK:1
> TSF:CHKUPL:FAIL
> !TSM:UPL:FAIL
> TSM:FPAR
> TSM:FPAR:STATP
> TSM:ID
> TSM:ID:OK,ID=10
> TSM:UPL
> TSF:PING:SEND,TO=0
> !TSF:MSG:SEND,10-10-0-0,s=255,c=3,t=24,pt=1,l=1,sg=0,ft=7,st=NACK:1
> TSF:CHKUPL:FAIL
> !TSM:UPL:FAIL
> TSM:FPAR
> TSM:FPAR:STATP
> TSM:ID
> TSM:ID:OK,ID=10
> TSM:UPL
> TSF:PING:SEND,TO=0
> !TSF:MSG:SEND,10-10-0-0,s=255,c=3,t=24,pt=1,l=1,sg=0,ft=8,st=NACK:1
> TSF:CHKUPL:FAIL
> !TSM:UPL:FAIL
> TSM:FPAR
> TSM:FPAR:STATP
> TSM:ID
> TSM:ID:OK,ID=10
> TSM:UPL
> TSF:PING:SEND,TO=0
> !TSF:MSG:SEND,10-10-0-0,s=255,c=3,t=24,pt=1,l=1,sg=0,ft=9,st=NACK:1
>

mahmoud-samy, 2016-07-27:
Gateway:

```
#define MY_GATEWAY_SERIAL
#define MY_NODE_ID 0
#define MY_DEBUG 
#define MY_RADIO_NRF24
#define MY_RF24_PA_LEVEL  RF24_PA_LOW
```

> 0;255;3;0;9;Starting gateway (RNNGA-,0;255;3;0;9;Starting gateway (RNNGA-, 2.0.0-beta)
> 0;255;3;0;9;TSM:INIT
> 0;255;3;0;9;TSM:INIT:TSP OK
> 0;255;3;0;9;TSM:INIT:GW MODE
> 0;255;3;0;9;TSM:READY
> 0;255;3;0;14;Gateway startup complete.
> 0;255;0;0;18;2.0.0-beta
> 0;255;3;0;9;No registration required
> 0;255;3;0;9;Init complete, id=0, parent=0, distance=0, registration=1
>

mahmoud-samy, 2016-07-27:
After switching to https://github.com/tekka007/MySensors/tree/TransportUpdate

Node:
> Starting sensor (RNNNA-, 2.0.0-beta)
> TSM:INIT
> TSM:INIT:TSP OK
> TSM:INIT:STATID,ID=10
> TSF:ASID:OK,ID=10
> TSM:FPAR
> TSM:FPAR:STATP
> TSM:ID
> TSM:ID:OK,ID=10
> TSM:UPL
> TSF:PING:SEND,TO=0
> !TSF:MSG:SEND,10-10-0-0,s=255,c=3,t=24,pt=1,l=1,sg=0,ft=0,st=NACK:1
> TSF:CHKUPL:FAIL
> !TSM:UPL:FAIL
> TSM:FPAR
> TSM:FPAR:STATP
> TSM:ID
> TSM:ID:OK,ID=10
> TSM:UPL
> TSF:PING:SEND,TO=0
> !TSF:MSG:SEND,10-10-0-0,s=255,c=3,t=24,pt=1,l=1,sg=0,ft=1,st=NACK:1
> TSF:CHKUPL:FAIL
> !TSM:UPL:FAIL
> TSM:FPAR
> TSM:FPAR:STATP
> TSM:ID
> TSM:ID:OK,ID=10
> TSM:UPL
> TSF:PING:SEND,TO=0
> !TSF:MSG:SEND,10-10-0-0,s=255,c=3,t=24,pt=1,l=1,sg=0,ft=2,st=NACK:1
> TSF:CHKUPL:FAIL
> !TSM:UPL:FAIL
> TSM:FPAR
> TSM:FPAR:STATP
> TSM:ID
> TSM:ID:OK,ID=10
> TSM:UPL
> TSF:PING:SEND,TO=0
> !TSF:MSG:SEND,10-10-0-0,s=255,c=3,t=24,pt=1,l=1,sg=0,ft=3,st=NACK:1
> TSF:CHKUPL:FAIL
> !TSM:UPL:FAIL
> TSM:FPAR
> TSM:FPAR:STATP
> TSM:ID
> TSM:ID:OK,ID=10
> TSM:UPL
> TSF:PING:SEND,TO=0
> !TSF:MSG:SEND,10-10-0-0,s=255,c=3,t=24,pt=1,l=1,sg=0,ft=4,st=NACK:1
> TSF:CHKUPL:FAIL
> !TSM:UPL:FAIL
> TSM:FPAR
> TSM:FPAR:STATP
> TSM:ID
> TSM:ID:OK,ID=10
> TSM:UPL
> TSF:PING:SEND,TO=0
> !TSF:MSG:SEND,10-10-0-0,s=255,c=3,t=24,pt=1,l=1,sg=0,ft=5,st=NACK:1
> TSF:CHKUPL:FAIL
> !TSM:UPL:FAIL
> TSM:FPAR
> TSM:FPAR:STATP
> TSM:ID
> TSM:ID:OK,ID=10
> TSM:UPL
> TSF:PING:SEND,TO=0
> !TSF:MSG:SEND,10-10-0-0,s=255,c=3,t=24,pt=1,l=1,sg=0,ft=6,st=NACK:1
> TSF:CHKUPL:FAIL
[... 187 lines cut ...]
> !TSF:MSG:SEND,10-10-0-0,s=255,c=3,t=24,pt=1,l=1,sg=0,ft=11,st=NACK:1
> TSF:CHKUPL:FAIL
> !TSM:UPL:FAIL
> TSM:FPAR
> TSM:FPAR:STATP
> TSM:ID
> TSM:ID:OK,ID=10
> TSM:UPL
> TSF:PING:SEND,TO=0
> !TSF:MSG:SEND,10-10-0-0,s=255,c=3,t=24,pt=1,l=1,sg=0,ft=12,st=NACK:1
> 

Gateway:

> 0;255;3;0;9;Starting gateway (RNNGA-=1
> 0;255;3;0;9;Starting gateway (RNNGA-, 2.0.0-beta)
> 0;255;3;0;9;TSM:INIT
> 0;255;3;0;9;TSM:INIT:TSP OK
> 0;255;3;0;9;TSM:INIT:GW MODE
> 0;255;3;0;9;TSM:READY
> 0;255;3;0;14;Gateway startup complete.
> 0;255;0;0;18;2.0.0-beta
> 0;255;3;0;9;No registration required
> 0;255;3;0;9;Init complete, id=0, parent=0, distance=0, registration=1
>

mahmoud-samy, 2016-07-27:
I forgot to mention that i'm using 
`#define MY_RF24_DATARATE 	RF24_1MBPS`
because I'm using nRF24l01 not nRF24l01+. and everything were working fine with 1.5.4

tekka007, 2016-07-27:
Please verify your radio settings on GW and node, eventually you have interferences or power issues. What you describe is not directly connected to the find parent issue, rather a radio link problem.

mahmoud-samy, 2016-07-28:
@tekka007 it's the radio. Replacing the nRF24l01 by nRF24l01+ solved the issue.

PR #522 (2016-07-28): Updates to MyTransport
- Fix issue with static parent assignment, this fixes #509 
- Fix GW callback
- Various minor code optimizations
- Update and unify transport log messages
- Doxygenize log messages
closed 2016-07-28 by henrikekblad