mysensors/MySensors #511: TSP:CHKUPL:FAIL error since June 26th commit

emc2cube, 2016-07-13:
Since June 26th commit all nodes end up in a boot loop, restarting every few seconds after
`TSP:CHKUPL:FAIL
!TSM:UPL:FAIL`

Reverting back to June 25th commit "fix" it.

Tested with Pro Mini 8Mhz and ESP8266 (as a node not a GW)

Controller: domoticz, last beta
GW: NRF24L01 on Pi GPIO ports
All nodes have fixed ID and parent ID is fixed to 0 (no repeaters yet)

tekka007, 2016-07-13:
Did you update GW and nodes with the 2.0.0 release? If yes, please provide a full debug log of both, GW and node when the reboot loop happens.

emc2cube, 2016-07-13:
Node on 2.0 release give the same error.
Gateway is https://github.com/mysensors/Raspberry/tree/master is there an update for this guy that I'm unaware of?

Log of a sensor on 2.0 release.
Every other node on the working June 25th beta version are off.

Starting repeater (RNNRE-, 2.0.0)
TSM:INIT
TSM:RADIO:OK
TSP:ASSIGNID:OK (ID=254)
TSM:FPAR
TSP:MSG:SEND 254-254-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
TSP:MSG:READ 0-0-254 s=255,c=3,t=8,pt=1,l=1,sg=0:0
TSP:MSG:FPAR RES (ID=0, dist=0)
TSP:MSG:FPAR (PPAR FOUND)
TSP:MSG:PAR OK (ID=0, dist=1)
TSM:FPAR:OK
TSM:ID
TSM:CHKID:OK (ID=254)
TSM:UPL
TSP:PING:SEND (dest=0)
TSP:MSG:SEND 254-254-0-0 s=255,c=3,t=24,pt=1,l=1,sg=0,ft=0,st=ok:1
TSP:MSG:READ 253-253-254 s=255,c=3,t=8,pt=1,l=1,sg=0:1
TSP:MSG:FPAR RES (ID=253, dist=1)
TSP:CHKUPL:FAIL (hops=255)
!TSM:UPL:FAIL
TSM:FPAR
TSP:MSG:SEND 254-254-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
TSP:MSG:READ 253-253-254 s=255,c=3,t=8,pt=1,l=1,sg=0:1
TSP:MSG:FPAR RES (ID=253, dist=1)
TSP:MSG:PAR OK (ID=253, dist=2)
TSP:MSG:READ 0-0-254 s=255,c=3,t=8,pt=1,l=1,sg=0:0
TSP:MSG:FPAR RES (ID=0, dist=0)
TSP:MSG:FPAR (PPAR FOUND)
TSP:MSG:PAR OK (ID=0, dist=1)
TSM:FPAR:OK
TSM:ID
TSM:CHKID:OK (ID=254)
TSM:UPL
TSP:PING:SEND (dest=0)
TSP:MSG:SEND 254-254-0-0 s=255,c=3,t=24,pt=1,l=1,sg=0,ft=0,st=ok:1
TSP:CHKUPL:FAIL (hops=255)
!TSM:UPL:FAIL
TSM:FPAR
TSP:MSG:SEND 254-254-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
TSM:FPAR
TSP:MSG:SEND 254-254-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
TSP:MSG:READ 253-253-254 s=255,c=3,t=8,pt=1,l=1,sg=0:1
TSP:MSG:FPAR RES (ID=253, dist=1)
TSP:MSG:PAR OK (ID=253, dist=2)
TSP:MSG:READ 0-0-254 s=255,c=3,t=8,pt=1,l=1,sg=0:0
TSP:MSG:FPAR RES (ID=0, dist=0)
TSP:MSG:FPAR (PPAR FOUND)
TSP:MSG:PAR OK (ID=0, dist=1)
TSM:FPAR:OK
TSM:ID
TSM:CHKID:OK (ID=254)
TSM:UPL
TSP:PING:SEND (dest=0)
TSP:MSG:SEND 254-254-0-0 s=255,c=3,t=24,pt=1,l=1,sg=0,ft=0,st=ok:1
TSP:CHKUPL:FAIL (hops=255)
!TSM:UPL:FAIL
TSM:FPAR
etc...

Gateway don't see the sensor / nothing in the logs (domoticz or using screen /dev/ttyMySensorsGateway)

tekka007, 2016-07-13:
Ok, RPI GW port has not been updated to support MySensors 2.0.0.
Issue has been opened on corresponding repository.

closed 2016-07-13 by tekka007

emc2cube, 2016-07-13:
I guess the fact it worked 2 weeks ago on 2.0.0-beta was quite confusing.
Thanks for looking into that!

Gohper, 2016-07-28:
New to this, but I think I have the same problem. I have built the serial gateway and the relay actuator. I see this on the later one. But It looks like it sends something four times.
On the gateway I get:
`0;255;3;0;9;TSP:SANCHK:OK
0;255;3;0;9;TSP:MSG:READ 255-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0:
0;255;3;0;9;TSP:MSG:BC
0;255;3;0;9;TSP:MSG:FPAR REQ (sender=255)
0;255;3;0;9;TSP:CHKUPL:OK
0;255;3;0;9;TSP:MSG:GWL OK
0;255;3;0;9;TSP:MSG:SEND 0-0-255-255 s=255,c=3,t=8,pt=1,l=1,sg=0,ft=0,st=bc:0
0;255;3;0;9;TSP:MSG:READ 255-255-0 s=255,c=3,t=3,pt=0,l=0,sg=0:
255;255;3;0;3;
0;255;3;0;9;TSP:MSG:READ 255-255-0 s=255,c=3,t=3,pt=0,l=0,sg=0:
255;255;3;0;3;
0;255;3;0;9;TSP:MSG:READ 255-255-0 s=255,c=3,t=3,pt=0,l=0,sg=0:
255;255;3;0;3;
0;255;3;0;9;TSP:MSG:READ 255-255-0 s=255,c=3,t=3,pt=0,l=0,sg=0:
255;255;3;0;3;`
And on the actuator:
`0;255;3;0;9;TSP:SANCHK:OK
0;255;3;0;9;TSP:MSG:READ 255-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0:
0;255;3;0;9;TSP:MSG:BC
0;255;3;0;9;TSP:MSG:FPAR REQ (sender=255)
0;255;3;0;9;TSP:CHKUPL:OK
0;255;3;0;9;TSP:MSG:GWL OK
0;255;3;0;9;TSP:MSG:SEND 0-0-255-255 s=255,c=3,t=8,pt=1,l=1,sg=0,ft=0,st=bc:0
0;255;3;0;9;TSP:MSG:READ 255-255-0 s=255,c=3,t=3,pt=0,l=0,sg=0:
255;255;3;0;3;
0;255;3;0;9;TSP:MSG:READ 255-255-0 s=255,c=3,t=3,pt=0,l=0,sg=0:
255;255;3;0;3;
0;255;3;0;9;TSP:MSG:READ 255-255-0 s=255,c=3,t=3,pt=0,l=0,sg=0:
255;255;3;0;3;
0;255;3;0;9;TSP:MSG:READ 255-255-0 s=255,c=3,t=3,pt=0,l=0,sg=0:
255;255;3;0;3;`
I am not sure about the id (255) is correct, as I am new to this

Gohper, 2016-07-29:
I think this is some electric thing. I uploaded serial gateway to the "actuator uno" and vice versa, it worked! At least as far as I can see... .  Maybe I need a capacitor on the power of the radio.
Or as I plan a "big" 3.3V regulator. Now I just need to figure out how to send commands via a terminal...

Gohper, 2016-07-30:
Hmm, cleared the eprom, getting the same error, maybe I was tired, and too many whiskey's...
But still it could be the transiecer power. But I don't know for sure.

Gohper, 2016-08-01:
Well I assigned a node id to the actuator and no error appears. I will check that this was the only changes I made.