mysensors/MySensors #516: esp8266 gateway will always start with DHCP

yoram-s, 2016-07-14:
current code implementation will begin the wifi before the config has taken an affect. 
MyGatewayTransportEthernet.cpp 
`
		(void)WiFi.begin(MY_ESP8266_SSID, MY_ESP8266_PASSWORD);
		#ifdef MY_IP_ADDRESS
			WiFi.config(_ethernetGatewayIP, _gatewayIp, _subnetIp);
		#endif
`
The call should be reverse, first the config then the begin.

tekka007, 2016-07-14:
Not really a bug, see the 4th paragraph in [WiFiConfig](https://www.arduino.cc/en/Reference/WiFiConfig):

> You can call WiFi.config() after WiFi.begin(), but the shield will initialize with begin() in the default DHCP mode. Once the config() method is called, it will change the network address as requested.

So, the current setup should also be fine, nevertheless, an upcoming PR will be addressed this.

Thanks for reporting.

tekka007, 2016-07-14:
Please, feel free to test and provide feedback: https://github.com/tekka007/MySensors/tree/ChangeInitOrder

PR #517 (2016-07-17): Calling WiFi.config() before WiFi.begin()
Calling WiFi.config() before WiFi.begin() forces begin() to configure the WiFi shield with the network addresses specified in config().

- Fixes #516
closed 2016-07-17 by fallberg