mysensors/MySensors #518: Serial port closed when using MySensors gateway with Domoticz

wmarkow, 2016-07-15:
I've started to play around with MySensors library and Domoticz home automation system. I had some issues with intergating the serial gateway with Domoticz. It took me a while to figure it out so I would like to share what I have found.

I have compiled and installed Domoticz on my local computer (I'm working on Ubuntu 15.10). Then I have uploaded the serial gateway sketch to Arduino UNO and configured MySensor Gateway USB in Domoticz to work with /dev/ttyUSB) serial port. The communication between Domoticz and the gateway was pretty good. Then I stared to play with Arduino sketches; made some modifications, checks and so on (I wanted to build a serial gateway with some mocked sensors to see how it looks in Domoticz). Then the weird things started to happen with the serial port connection. Here is a snippet of Domoticz logs:

```
2016-07-15 12:57:18.745  MySensors: Using serial port: /dev/ttyUSB0
2016-07-15 12:57:18.794  Error: Serial Port closed!... Error: End of file
2016-07-15 12:57:19.793  MySensors: retrying in 30 seconds...
2016-07-15 12:57:48.797  MySensors: Using serial port: /dev/ttyUSB0
```

or

```
2016-07-15 12:57:18.745  MySensors: Using serial port: /dev/ttyUSB0
2016-07-15 12:57:18.793  MySensors: NodeID: 0, ChildID: 7, MessageType: 0, Ack: 0, SubType: 6, Payload: 10
2016-07-15 12:57:18.794  Error: Serial Port closed!... Error: End of file
2016-07-15 12:57:19.793  MySensors: retrying in 30 seconds...
2016-07-15 12:57:48.797  MySensors: Using serial port: /dev/ttyUSB0
```

After some time of investigations I gave up and switched off my computer. The next day I wanted to make some new investigations on this issue. I switched my laptop on, plugged Arduino (my sensor gateway) into USB port and started Domoticz to see if the error still takes place... and what did I see? The serial port communication works just fine. Ok then - I have started to work with Arduino code again and uploaded some sketches into it, and the serial communications started to be broken again. I t took me a while to figure it out that:
** uploading the hex to Arduino with avrdude somehow "breaks" the serial port so Domoticz can not use it correctly**
** restarting my linux machine fixes the serial port communication **

Good thing is that I do not have to restart my computer to fix the serial port. On linux I can reload the usbserial driver and Domoticz starts working fine again. Thanks to [this post](http://superuser.com/a/596911).

tbowmo, 2016-07-15:
Do you shut down domoticz when uploading new sketches? Could be the problem if you don't, since both avrdude and domoticz tries to use the same serial port at the same time. 

Another thing, what does dmesg tell you after an upload? Is the USB device re-enumerating on the USB bus? Could it get a new dev identifier (/dev/usbTTY1 instead of 0).

I don't think that this is a problem with mysensors, but an issue with serial devices on Linux

wmarkow, 2016-07-15:
Yes - I did shut down domoticz when uploading the sketches. Looks like the device is not re-enumerating after uploading. Most of the times it was still the same /dev/ttyUSB0. I had like two cases when it re-enumerates to /dev/ttyUSB1 but I did it in purpose to see if the device after re-enumerating will work (however domoticz didn't work).

I know that it is not related to mysensors itself but - like you said - with serial devices on Linux. I have noticed this issue when starting to play with MySensors and Domoticz so I decided to rise an issue here in case if someone have similar issues in the future.

closed 2016-07-31 by henrikekblad