mysensors/MySensors #523: MySensors via XBee??

MYeager1967, 2016-07-19:
How much work would it be to use XBee radios with this? I have a fairly decent sensor network that I wouldn't mind converting so long as it allows control signals to be sent back. I'm transitioning from a home brew automation system to a Vera Plus and this might make including my current sensors and devices much easier.

fallberg, 2016-07-19:
I do not know of any core team member with a working xbee setup to test such a thing. But if you have a working setup, feel free to submit a pr or reference a open source project and we can at least gave a look at it.

henrikekblad, 2016-07-31:
I'm closing this as XBee support won't happen in a near future.

closed 2016-07-31 by henrikekblad