mysensors/MySensors #527: nrf24l01 radios aren't working with v2.0.x anymore

mahmoud-samy, 2016-07-28:
I used to work with nrf24l01 (not nrf24l01+).
Just by setting nrf datarate by 1Mbps, it was working fine with mySensors v1.x.

After migration to v2.0.x, my gw and all nodes stop working (more than 20 node). By replacing nrf24l01 radio by nrf24l01+ everything back to life.

tekka007, 2016-07-28:
Please enable MY_DEBUG_VERBOSE_RF24 and post the debug log from GW and node here - this will help to nail down the issue. Thanks

mahmoud-samy, 2016-07-28:
OK, I just need some time to do that.

tekka007, 2016-07-29:
Thanks for the info - that's helpful for further investigations. 
There is one thing that came to my mind: Did you try if a mixed setup, i.e. nRF24L01 on GW and nRF24L01P on node and vice versa is working (@1Mbps)?

Thanks again for your valuable help.

mahmoud-samy, 2016-07-29:
Gateway: nRF24L01  / Node: nRF24L01+
https://gist.github.com/mahmoud-samy/51e379b120dea75a761ffa250139f283
Gateway: nRF24L01+ / Node: nRF24L01
https://gist.github.com/mahmoud-samy/5641130cdaf70a23ae7af5a792a28bfc

tekka007, 2016-07-30:
Great! I'll have a look at the logs - what surprises me, is that your nRF24L01  (non-P) radio nodes worked for 1.5.4, since there is a non-P version check during initialisation.

tekka007, 2016-07-30:
Please test this version [here](https://github.com/tekka007/MySensors/tree/nRF24L01), ideally again with reporting the verbose RF24 log in non-P / mixed / and P setups. Thanks!

mahmoud-samy, 2016-07-31:
@tekka007 you can forget about v1.5.x since I was working on a special branch with a lot of modifications and hacks not the original one. After switching to the original 1.5.0, 1.5.1, 1.5.4 and disabling isPVariant check, non of nodes work, same as my case with v2.0.x

My special branch was branched out of MySensors v1.3 with a lot of customization and cherry-picks to match 1.5. BTW I don't want to continue working on that branch anymore, and stick to the original implementation. 

Soon, I'll test your new changes. 
Thanks :+1:

mahmoud-samy, 2016-07-31:
Regarding your changes [here](https://github.com/tekka007/MySensors/commit/68211fff85360344633d998682a354c3cdc971ba). Please note that MY_RF24_SANITY_CHECK symbol is NOT defined in my case

mahmoud-samy, 2016-07-31:
@tekka007 please check the results [here](https://gist.github.com/mahmoud-samy/cd7fa53cfdcb799bffd3b9ca1c03f097)

mahmoud-samy, 2016-07-31:
@tekka007 [here](https://gist.github.com/mahmoud-samy/e2fa19dedba11af346020fa155d30abd) you will find the results of experiment 3 after running for a longer duration.

tekka007, 2016-07-31:
@mahmoud-samy Perfect, the test results look promising. I did one additional change ([same repo](https://github.com/tekka007/MySensors/tree/nRF24L01)) - this should fix the issues with non-P modules. Can you confirm that non-P modules (at 1Mbps) are working with this solution?

mahmoud-samy, 2016-08-01:
https://gist.github.com/mahmoud-samy/4da161bc0a75c1551dd0e45b7e0edc42

phuckhoi2494, 2016-08-01:
Hello Mahmoud-samy and tekka007, I'm working with version 2.0 to build a Gateway and a Sensor node using UNO and Pro mini. I see this on Serial monitor:
0;255;3;0;9;Starting gateway (RNNGA-, 2.0.0)
0;255;3;0;9;TSM:INIT
0;255;3;0;9;TSM:RADIO:OK
0;255;3;0;9;TSM:GW MODE
0;255;3;0;9;TSM:READY
0;255;3;0;14;Gateway startup complete.
0;255;0;0;18;2.0.0
0;255;3;0;9;No registration required
0;255;3;0;9;Init complete, id=0, parent=0, distance=0, registration=1
0;255;3;0;9;TSP:MSG:READ 255-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0:
0;255;3;0;9;TSP:MSG:BC
0;255;3;0;9;TSP:MSG:FPAR REQ (sender=255)
0;255;3;0;9;TSP:CHKUPL:OK (FLDCTRL)
0;255;3;0;9;TSP:MSG:GWL OK
0;255;3;0;9;TSP:MSG:SEND 0-0-255-255 s=255,c=3,t=8,pt=1,l=1,sg=0,ft=0,st=bc:0
0;255;3;0;9;TSP:MSG:READ 255-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0:
0;255;3;0;9;TSP:MSG:BC

I want to send a random number from the node to the gateway, could you help me to do this ? 
Thank you very much

tekka007, 2016-08-01:
@phuckhoi2494 Welcome to MySensors, this is the wrong place for your request. Please have a look at the mysensors forum and submit a post in the appropriate category. Thanks.

tekka007, 2016-08-01:
@mahmoud-samy Thanks for your help with testing this PR. The logs confirm that this fixes the issue.

mahmoud-samy, 2016-08-01:
@tekka007 I see.

Thank you for your efforts, it's really appreciated.

PR #533 (2016-08-01): Fix compatibility issue with nRF clones, update documentation
- nRF24: Fix #527 
- Doxygen documentation: Add table describing transport debug log messages
- Fix typos
closed 2016-08-01 by henrikekblad