mysensors/MySensors #528: Problem with sleep/interrupts w/atmega328p

mannkind, 2016-07-29:
I have a node that sleeps until interrupted on digital pin 2, sends the state of the digital pin 2, then goes back to sleep until interrupted again.  I'm utilizing the MySensors `sleep(INTERRUPT, CHANGE)` function at the end of the `loop()`

Most recently, I was using commit 312a69b and everything was working as expected. Tonight, because I was bored, I started to upgrade to 929b0013 (the latest commit in development as of writing this). Immediately I noticed an issue – the node was no longer sleeping until interrupt and was rapidly sending the state of the pin over and over and over again. 

Checking the commit history, I see there are two commits related to interrupts: 1305aec and 83cbb92. The problem seems to have been introduced in commit 1305aec. I can revert from 312a69b - f0f2542d (the one just before 1305aec) and the issue goes away.

Yveaux, 2016-07-29:
Main change in this changeset is level changes on the interrupt pin could possible be missed in teh old code. The changes suppress interrupts until going to sleep. If an interrupt happens inside the sleep() function call it should now be detected at all times.
Could it be that in tyour case the interrupt fires immediately when trying to go to sleep (that would previously be ignored)?
Can you provide an exact description of your test setup; e.g. what is used to trigger the interrupt etc.?

mannkind, 2016-07-29:
The interrupt is physically triggered by a normally open w/magnet reed switch - magnet present and the circuit is open, magnet taken away the circuit is closed. 

The pin is pulled high in `setup()`

In case the reed switch was coincidentally going bad/triggering the interrupt like you mentioned, I removed it from the circuit and used jumper-wire to close/open the circuit.  The problem still persists.

mannkind, 2016-07-29:
This sketch (based on one of the examples) replicates the problem.
```
// Based on https://raw.githubusercontent.com/mysensors/MySensors/development/examples/BinarySwitchSleepSensor/BinarySwitchSleepSensor.ino
#define MY_RADIO_NRF24
#include <SPI.h>
#include <MySensors.h>

MyMessage msg(0, V_TRIPPED);

void setup() {
  pinMode(2, INPUT_PULLUP);
}

void presentation() {
  present(0, S_DOOR);
}

void loop() {
  uint8_t value = digitalRead(2);
  send(msg.set(value==HIGH));

  sleep(0, CHANGE);
}

Yveaux, 2016-07-29:
Both a reed switch and jumper wire will bounce when closing the connection. Effectively the signal will rapidly toggle between high /low multiple times for a short period of time. If your sensor goes to sleep while the contacts are still bouncing, it could immediately wake again. 
Try with a delay (e.g. 1 sec) before going to sleep and see if that changes its behavior.

mannkind, 2016-07-29:
I do know about the denouncing and technically already have a "delay"… though I'm not using `delay(unsigned long ms)` but another call to `sleep(unsigned long ms)`. Maybe that's the issue? I've been doing that since before MySensors, when I used the Rocket Scream Low Power lib because these specific sensors are battery powered.

Here is a _minimal_ amount of code that demonstrates the issue: http://pastebin.com/MxaS4Qje

I uploaded it to my node with an ID of 1.

… using MySensors 312a69b
```
mysensors_rx/1/255/0/0/17 2.0.0-beta
mysensors_rx/1/255/3/0/6 0
mysensors_rx/1/0/0/0/0 (null)
mysensors_rx/1/0/1/0/16 1
<end>
```

… using MySensors 929b001 
```
mysensors_rx/1/255/0/0/17 2.0.1-beta
mysensors_rx/1/255/3/0/6 0
mysensors_rx/1/0/0/0/0 (null)
mysensors_rx/1/0/1/0/16 1
mysensors_rx/1/0/1/0/16 1
mysensors_rx/1/0/1/0/16 1
mysensors_rx/1/0/1/0/16 1
mysensors_rx/1/0/1/0/16 1
mysensors_rx/1/0/1/0/16 1
mysensors_rx/1/0/1/0/16 1
mysensors_rx/1/0/1/0/16 1
mysensors_rx/1/0/1/0/16 1
mysensors_rx/1/0/1/0/16 1
mysensors_rx/1/0/1/0/16 1
mysensors_rx/1/0/1/0/16 1
mysensors_rx/1/0/1/0/16 1
mysensors_rx/1/0/1/0/16 1
mysensors_rx/1/0/1/0/16 1
mysensors_rx/1/0/1/0/16 1
mysensors_rx/1/0/1/0/16 1
… 
<until I pull the power>
```

Yveaux, 2016-07-30:
Apart from the extra sleep the sketch is more or less identical to what I tested with. 
Busy right now, but I will try to reproduce to find out what causes the issue.

mannkind, 2016-07-30:
No problem!  I have potential work-arounds now, so it's not a huge issue; it's just weird. :)

Let me know if I can test anything for you, provide logs, etc etc. I'm happy to help, but I won't be able to do so before sometime Sunday evening (PDT).

Yveaux, 2016-07-31:
Built a setup (Your example sketch, ProMini 8Mhz, ESP8266 Gateway, Arduino 1.6.9, MySensors 929b001) and tried alternately shorting IRQ0 to GND an disconnecting it.
Works just fine, with and without sleep. It only generates a single high/low transition (with sleep) or 2 high/low transitions without (what is to be expected due to bouncing of the contact).
Only conclusion I have is that it must be hardware related somehow.
If you have access to an oscilloscope you could connect it to see what's really going on on the IRQ pin.
As a workaround you could keep the extra sleep() in.

closed 2016-07-31 by Yveaux

mannkind, 2016-07-31:
No oscilloscope; I'm not that sophisticated. :)

I think I'm confused - the *extra* sleep is what seems to be causing the issue. If I remove it, or replace it with delay, I get the expected behavior. 

Well at least there will be this issue to reference *if* others experience trouble come the release of 2.0.1.