mysensors/MySensors #529: using my own wifi client.

Br3nda, 2016-07-30:
I was using MySensors 1.5 on esp8266 with WifiManager to set the ssid and pass. 
Is this possible in 2.0? I've tried several things, but can't find a way to manage the wifi client myself instead of the mysensors library taking over. It get stuck at build on:
```
MyGatewayTransportMQTTClient.cpp:201:20: error: 'MY_ESP8266_SSID' was not declared in this scope
   (void)WiFi.begin(MY_ESP8266_SSID, MY_ESP8266_PASSWORD);
```

henrikekblad, 2016-07-30:
For MQTT or regular Ethernet? 

As you can see the WiFi.begin gets called by the library.. Why do you need to manage it yourself? 

Add 
```
#define MY_ESP8266_SSID "<your-ssid>"
#define MY_ESP8266_PASSWORD "<your-pw>"
```
In the sketch before including MySensors.h.

If you need to initialise things before the library, create a ```void before() {}``` function in the sketch and do the inits there.

Br3nda, 2016-07-31:
I need to set the ssid and pw at runtime.
not at #define time.

Is this possible any more?

Br3nda, 2016-07-31:
to be clearer:
The sensors are going places where:
1) we don't know the ssid/pass
2) the ssid/pass can change over time.

In order re-configure the ssid/pass in the field, we have been using WifiManager. It make the ESP8266 appear as an AP, then serves up a webpage where you can tell it the ssid. It saves this then goes back to being a MySensors gateway.

Hence: we need to set the ssid/pass at runtime.

I am trying to upgrade to 2.0 and getting stuck.

trlafleur, 2016-07-31:
A #define just set a name to another name.  

Remove the #define and just set the variable to what you want at run time. 

Basic 'C'

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

Br3nda, 2016-07-31:
duh!

:-) okay - thank i was just realising i could put my own method there.

Thanks

closed 2016-07-31 by Br3nda

bkrajendra, 2018-01-01:
I was also trying to add WiFi SmartConfig in esp8266 gateway but unable to do it. tried putting it in before function but it does not work as expected.

Br3nda, 2018-01-03:
@bkrajendra have a look at the code here https://github.com/WhareHauora/wharehauora-wifi-gateway/

bkrajendra, 2018-01-03:
@Br3nda Thanks a lot..
It really helped me. Now trying to add SmartConfig.