mysensors/MySensors #562: LEDs won't work when MY_DEFAULT_xx_LED_PIN is not defined

Yveaux, 2016-08-17:
When a sketch defines MY_LEDS_BLINKING_FEATURE the MySensors library assumes MY_DEFAULT_xxx_LED_PIN are also defined.
When they are not, and MY_HW_xxx_LED_PIN are also not defined (https://github.com/mysensors/MySensors/blob/development/MySensors.h#L75) the LED blinking will silently fail.

Furthermore, the library assumes that if one of the MY_DEFAULT_xxx_LED_PIN are defined, that all 3 LEDs are present and configured (https://github.com/mysensors/MySensors/blob/development/core/MyLeds.cpp#L40).
This may not be the case, and it should be possible to have e.g. only an error LED.

Yveaux, 2016-08-19:
Conclusion after discussing:
- MY_HW_xxx_LED_PIN can be defined in variants (board specific) to automatically assign LED pins
- User should define any of MY_DEFAULT_xxx_LED_PIN in his sketch, when he wants to assign LEDs to MySensors
- MY_LEDS_BLINKING_FEATURE will be depricated, as defining one or more of MY_DEFAULT_xxx_LED_PIN will automatically enable LED blinking by the library.
- Example Gateway sketches must be extended with MY_DEFAULT_xxx_LED_PIN definitions, and MY_LEDS_BLINKING_FEATURE  should be removed (**add conversion notes to release notes for 2.0.1**)
- Refrain from falling back to some default led pins, when MY_DEFAULT_xxx_LED_PIN are not defined
- Individual LEDs can be enabled by defining just the MY_DEFAULT_xxx_LED_PIN - no longer all or nothing!

GertSanders, 2016-08-19:
Like !

PR #566 (2016-08-20): Fixes LED issues discussed in #562
Removing MY_LED_BLINKING_FEATURE, as we now are
using MY_DEFAULT_LED_xx_PIN defines to enable
LED blinking feature.

updated Gateway sketches to not refer to MY_LED_BLINKING_FEATURE any more

fixes #562
commit 15b46f80 (2016-08-20): Fixes LED issues as discussed in #562

Removing MY_LED_BLINKING_FEATURE, as we now are
using MY_DEFAULT_LED_xx_PIN defines to enable
LED blinking feature.

updated Gateway sketches to not refer to
MY_LED_BLINKING_FEATURE any more

fixes #562
closed 2016-08-20 by henrikekblad