mysensors/MySensors #563: Static IP Address issue on GatewayESP8266MQTTClient

brmo, 2016-08-18:
Defining MY_IP_ADDRESS for static does not connect to MQTT server. Also still states dhcp client start in debug log. If i comment out the MY_IP_ADDRESS line, it connects to DHCP and MQTT just fine.

Version 2.0 release

---start of log for trying to use static ip----

0;255;3;0;9;Starting gateway (RNNGE-, 2.0.0)
0;255;3;0;9;TSM:INIT
0;255;3;0;9;TSM:RADIO:OK
0;255;3;0;9;TSM:GW MODE
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
cnt 

connected with NachoIOT, channel 1
dhcp client start...
0;255;3;0;9;TSM:READY
f r0, scandone
...................pm open,type:2 0
..........................................................................................................................................................................................................................................................................................................................

---end of log for trying to use static ip----

---start of log for trying to use dhcp----

0;255;3;0;9;Starting gateway (RNNGE-, 2.0.0)
0;255;3;0;9;TSM:INIT
0;255;3;0;9;TSM:RADIO:OK
0;255;3;0;9;TSM:GW MODE
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
cnt 

connected with NachoIOT, channel 1
dhcp client start...
0;255;3;0;9;TSM:READY
f r0, scandone
...ip:10.200.62.109,mask:255.255.255.0,gw:10.200.62.254
.IP: 10.200.62.109
0;255;3;0;9;No registration required
0;255;3;0;9;Init complete, id=0, parent=0, distance=0, registration=1
IP: 10.200.62.109
0;255;3;0;9;Attempting MQTT connection...
0;255;3;0;9;MQTT connected

---end of log for trying to use dhcp----

tekka007, 2016-08-19:
@brmo I cannot reproduce what you describe.
Please indicate Arduino IDE/ESP board def version and upload the sketch you are using.

brmo, 2016-09-05:
Arduino 1.6.11, Wemos D1 Mini, and the sketch is the example in the [project](https://github.com/mysensors/MySensors/blob/development/examples/GatewayESP8266MQTTClient/GatewayESP8266MQTTClient.ino) only changing the SSID/Password, MQTT Controller IP and the Static IP/Gateway/Subnet for my network.

mohsh86, 2016-09-15:
am having the same issue and it occurs randomly, sometimes it gets the static IP and it connects to the MQTT broker, most of the times it has the behavior described above.

using Wemos D1 Mini with RFM69HW (having another radio issue as well)

RubenKelevra, 2016-09-20:
I think we discovered the same issue with W5100 ethernet module ...

Did someone bisect already which commit broke it?

RubenKelevra, 2016-09-20:
This bug seems to be related to this commit, it changes the dhcp / static-ip config. But I didn't found the bug while reading the commit.

https://github.com/mysensors/MySensors/commit/77bb94c16f01fbd1884165d034b2eaffc0a12f8a

RubenKelevra, 2016-09-20:
@brmo do you get time to test this?

brmo, 2016-09-20:
@RubenKelevra What would you like me to test?  It sounds like from your statements, this hasnt been fixed?

RubenKelevra, 2016-09-20:
@brmo you might want to check out the version one commit ahead of this and than this commit to verify that the issue is there. :)