mysensors/MySensors #567: S_RADIOACTIVITY

henrikekblad, 2016-08-20:
New device type for geiger counters. Sending in usv (uSv/hr) as V_LEVEL

usv = cpm * x;

x = tube constant 

https://forum.mysensors.org/topic/1444/geiger-counter/6

FotoFieber, 2016-08-21:
The clicks measured is, what the tube has registered. And this is what you hear. 😄  I really like to see this value too.

The usv number form above is quite small when measuring every minute. If I understand you right, you would multiply it by 60 to have a relative value per hour. Am I correct? For a UI I think this is the value you want to see. The question is, isn't this better done in the controller?

With V_LEVEL, you would average all of them them and multiply by 365 * 24 to get the total radiotion.

With the cpm, I can just sum them up in the time and multiply with the factor of the tube.

mickecamino, 2017-07-22:
I vote for V_LEVEL in uSv/h because that depends on the tube constant.
And I also vote for V_PULSECOUNT for cpm so that other calculations can be made.

joelwutzke, 2017-12-02:
I would love to see this implemented.

kisse66, 2018-03-30:
Another vote for both values

JanChrisKlein, 2019-01-06:
Hej,

what's the status here? I'd like to add a Geiger Counter to my Domoticz installation and using S_DUST and V_LEVEL seems silly …

I second the vote for cpm and uSv/h …

CHristoph