mysensors/MySensors #572: Using namespaces an .cpp files brings up errors when compiling

sidey79, 2016-08-23:
I'm trying to implement sensor specific code in seperate cpp and header files and then include them as needed.

But, i can't get it work, compiling this. Code is reduced to basics:

**tPWR.h**
```
#ifndef _TPWR_h
#define _TPWR_h

#define MY_CORE_ONLY
#include <MySensors.h>

namespace tPWR {
	#define CHILD_ID 1									// Id of the sensor child
	MyMessage mPOWERcnt(CHILD_ID, V_KWH);
	#define DIGITAL_INPUT_SENSOR 2						// The digital input you attached your sensor.  (Only 2 and 3 generates interrupt!)
	void setup();
}
#endif
```

**tPWR.cpp**
```
#include "tPWR.h"
namespace tPWR {
void setup()
	{
		pinMode(DIGITAL_INPUT_SENSOR, INPUT_PULLUP);
	}
}

```

I get an error like this:
```
tPWR.cpp.o (symbol from plugin)*: In function wakeUp()
(.text+0x0)*: multiple definition of wakeUp()
EthernetGateway.cpp.o (symbol from plugin)*: (.text+0x0): first defined here
```

My main.ino file looks like this
```
#include <MySensors.h>
#include "tPWR.h"

void setup()
{
	tPWR::setup();
}

```

Currently i have no idea how to  fix this.

henrikekblad, 2016-08-23:
Including MySensors.h drags in actual code. So it can only be include once. 

Can't you just include MyMessage.h in your tPWR.h?

sidey79, 2016-08-24:
Yes, this works if i move my complete defenition and declarion to the header file.

**tPWR.h**
'''
#ifndef _TPWR_h
#define _TPWR_h

#include "core/MyMessage.h"

namespace tPWR {
    #define CHILD_ID 1                                  // Id of the sensor child
    MyMessage mPOWERcnt(CHILD_ID, V_KWH);
    #define DIGITAL_INPUT_SENSOR 2                      // The digital input you attached your sensor.  (Only 2 and 3 generates interrupt!)
    void setup()     
    {
        pinMode(DIGITAL_INPUT_SENSOR, INPUT_PULLUP);
    };
}
#endif
```

tPWR.cpp has no entrys anymore.

**Thanks.**

closed 2016-08-24 by tekka007