mysensors/MySensors #575: analogRead() on ESP8266 is not working

marceloaqno, 2016-08-28:
ESP8266 has a single ADC channel available, it may be used either to read voltage at ADC pin, or to read internal voltage (VCC).
To read voltage and make use of hwCPUVoltage() function the following line is required:
ADC_MODE(ADC_VCC);
https://github.com/mysensors/MySensors/blob/development/core/MyHwESP8266.cpp#L117

but analogRead() won't work anymore (always returns 65535).

mfalkvidd, 2016-12-17:
According to https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/doc/reference.md#analog-input **ADC pin must be kept unconnected** for VCC reading to work. If that's correct, I suggest MySensors leaves the pin as regular analog pin, making the analog pin available to the sketch programmer. We can't know if the analog pin is kept unconnected, so reading VCC wonät be reliable anyway.

An alternative might be to switch into ADC_VCC mode briefly when taking VCC measurement, and then switching back to "normal" mode. Switching back should be possible by calling ADC_MODE(ADC_TOUT)?[1] but if the sketch programmer connected something to the pin the reading may be inaccurate.

1: https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/cores/esp8266/Esp.h#L68

marceloaqno, 2016-12-18:
Is there a dynamic way to change de ADC_MODE? I tried: http://www.esp8266.com/viewtopic.php?f=13&t=7908&start=4#p40194, but it didn't work.

sushukka, 2016-12-18:
I just faced this problem when trying to read ACS712 hall sensor on ESP8266 Wemos D1 Mini. Everything worked fine until I included the MySensors library. After that just 65536 values from A0. As the A0 is the only analog input on ESP8266 I'm wondering what is the reason to sacrifice it for VCC reading? 

However, hot it working by commenting the ADC_MODE line in MyHwESP8266.CPP. Thanks mfalkvidd for the help. :)

I also think that using some jumping between modes is not a good idea. It fits badly to the nature of analog reading. Moreover if you really should not have anything connected to A0 when using it as VCC reading, there are not many options but just to leave it alone for analogRead.

mfalkvidd, 2016-12-18:
Related pull request: https://github.com/mysensors/MySensors/pull/695

tekka007, 2016-12-18:
See #696, introduced _ESP8266_ADC_VCC_DISABLE_ switch if analogRead(A0) is used

closed 2016-12-22 by tekka007